OpenZWave/open-zwave #674: Segmentation fault with central scene devices

sarakha63, 2015-10-17:
Hi justin now tha i have all the logs i create the issue.

So as i said by mail, there is an issue with central scene devices. with a segmentation fault

here is the gdb traceback
```
#1  0xb6d6c082 in _IO_vfprintf_internal (s=s@entry=0xb4efdda0,
    format=format@entry=0xa10ac "Received Central Scene set from node %d: scene id=%d in %s seconds. Sending event notification.", ap=..., ap@entry=...)
    at vfprintf.c:1642
        len = <optimized out>
        string_malloced = <optimized out>
        step0_jumps = {0, -6120, -6064, -5970, -6016, -5826, -5748, -5286, -5048, -5632, -5560, -4868, -4660, -4604, -4308, -2442, -4188, -2592, -2218,
          -2212, -766, -20, -1798, -1008, -958, -408, -2002, -4430, -4374, -5688}
        space = <optimized out>
        is_short = <optimized out>
        use_outdigits = <optimized out>
        step1_jumps = {0, 0, 0, 0, 0, 0, 0, 0, 0, -5632, -5560, -4868, -4660, -4604, -4308, -2442, -4188, -2592, -2218, -2212, -766, -20, -1798, -1008,
          -958, -408, -2002, -4430, -4374, 0}
        group = <optimized out>
        prec = <optimized out>
        step2_jumps = {0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -5560, -4868, -4660, -4604, -4308, -2442, -4188, -2592, -2218, -2212, -766, -20, -1798, -1008, -958,
          -408, -2002, -4430, -4374, 0}
        string = <optimized out>
        left = 0
        is_long_double = <optimized out>
        width = 0
        step3a_jumps = {0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -4922, 0, 0, 0, -4308, -2442, -4188, -2592, -2218, 0, 0, 0, 0, -1008, 0, 0, 0, 0, 0, 0}
        alt = <optimized out>
        showsign = <optimized out>
        is_long = <optimized out>
        is_char = <optimized out>
        pad = 32 ' '
        step3b_jumps = {0 <repeats 11 times>, -4660, 0, 0, -4308, -2442, -4188, -2592, -2218, -2212, -766, -20, -1798, -1008, -958, -408, -2002, 0, 0, 0}
        step4_jumps = {0 <repeats 14 times>, -4308, -2442, -4188, -2592, -2218, -2212, -766, -20, -1798, -1008, -958, -408, -2002, 0, 0, 0}
        is_negative = <optimized out>
        number = {longlong = 1, word = 1}
        base = <optimized out>
        the_arg = {pa_wchar = 3035618912 L'\xb4efde60', pa_int = -1259348384, pa_long_int = -1259348384, pa_long_long_int = 3035618912,
          pa_u_int = 3035618912, pa_u_long_int = 3035618912, pa_u_long_long_int = 3035618912, pa_double = 1.4997950182851822e-314,
          pa_long_double = 1.4997950182851822e-314, pa_string = 0xb4efde60 "\374\003",
          pa_wstring = 0xb4efde60 L"ϼ\xb6f731cf\xb4efdec8\xc4ae4\xb4efde7c\x2f3af\xa10ac\x6f666e49‬\x65646f4e\xdd068\a\xa10ac\xc42b0\xb4efdec0\xb4efdec4\xb4efdec8\x2f595\xb4efe3e0\xb4efe401\xc42b0\xb4efded0\xb4efe7df", pa_pointer = 0xb4efde60, pa_user = 0xb4efde60}
        spec = <optimized out>
        _buffer = {__routine = 0xb6d6c562 <_IO_vfprintf_internal+8906>, __arg = 0xb6d6c562 <_IO_vfprintf_internal+8906>, __canceltype = -1227438750,
          __prev = 0xb6dff070 <step0_jumps>}
        _avail = <optimized out>
        thousands_sep = <optimized out>
        grouping = 0xffffffff <error: Cannot access memory at address 0xffffffff>
        done = <optimized out>
        f = 0xa10e5 "s seconds. Sending event notification."
---Type <return> to continue, or q <return> to quit---
        lead_str_end = <optimized out>
        end_of_spec = <optimized out>
        work_buffer = '\000' <repeats 32 times>, "\060", '\000' <repeats 35 times>, "\001\000\000\000 ", '\000' <repeats 15 times>, "x", '\000' <repeats 11 times>, "\300\337\357\264\000\000\000\000\377\377\377\377\002\000\000\000\000\000\000\000\377\377\377\377\374\020\t\000\000\000\000\000p\360߶\204\337\357\264\002\021\t\000\000\000\000\000\070\332\357\264\002", '\000' <repeats 11 times>, "\030\021\t\000\064\332\357\264\000 \342\266\000\000\000\000\036\021\t\000\264\337\357\264", '\000' <repeats 32 times>...
        workstart = <optimized out>
        workend = <optimized out>
        ap_save = {__ap = 0xb4efe724}
        nspecs_done = <optimized out>
        save_errno = <optimized out>
        readonly_format = <optimized out>
        args_malloced = <optimized out>
        specs = <optimized out>
        specs_malloced = <optimized out>
        jump_table = "\001\000\000\004\000\016\000\006\000\000\a\002\000\003\t\000\005\b\b\b\b\b\b\b\b\b\000\000\000\000\000\000\000\032\000\031\000\023\023\023\000\035\000\000\f\000\000\000\000\000\000\025\000\000\000\000\022\000\r\000\000\000\000\000\000\032\000\024\017\023\023\023\n\017\034\000\v\030\027\021\026\f\000\025\033\020\000\000\022\000\r"
        __PRETTY_FUNCTION__ = "_IO_vfprintf_internal"
#2  0xb6d861fc in _IO_vsnprintf (string=0xb4efded0 "Received Central Scene set from node 23: scene id=1 in ", maxlen=<optimized out>,
    format=0xa10ac "Received Central Scene set from node %d: scene id=%d in %s seconds. Sending event notification.", args=...) at vsnprintf.c:119
        sf = {f = {_sbf = {_f = {_flags = -72515583, _IO_read_ptr = 0xb4efded0 "Received Central Scene set from node 23: scene id=1 in ",
                _IO_read_end = 0xb4efded0 "Received Central Scene set from node 23: scene id=1 in ",
                _IO_read_base = 0xb4efded0 "Received Central Scene set from node 23: scene id=1 in ",
                _IO_write_base = 0xb4efded0 "Received Central Scene set from node 23: scene id=1 in ", _IO_write_ptr = 0xb4efdf07 "",
                _IO_write_end = 0xb4efe2cf "", _IO_buf_base = 0xb4efded0 "Received Central Scene set from node 23: scene id=1 in ",
                _IO_buf_end = 0xb4efe2cf "", _IO_save_base = 0x0, _IO_backup_base = 0x0, _IO_save_end = 0x0, _markers = 0x0, _chain = 0x0, _fileno = 0,
                _flags2 = 0, _old_offset = -1259348288, _cur_column = 0, _vtable_offset = -17 '\357', _shortbuf = "\264", _lock = 0x0,
                _offset = -5372735459793167448, _codecvt = 0x0, _wide_data = 0xffffffff, _freeres_list = 0x0, _freeres_buf = 0x13, _freeres_size = 1020,
                _mode = -1,
                _unused2 = "\304\336\357\264\310\336ﴄ\351\357\264C\332ض\204\351\357\264\000\000\000\000\023\000\000\000\000`\371\266\000`\371\266\235", <incomplete sequence \363\266>}, vtable = 0xb6e21ba8 <_IO_strn_jumps>}, _s = {_allocate_buffer = 0x0,
              _free_buffer = 0xb6f71bef <std::string::_Rep::_S_create(unsigned int, unsigned int, std::allocator<char> const&)+74>}},
          overflow_buf = "\006\000\000\000\267.\367\266\000\000\000\000|\336\357\264\310\336\357\264x\336\357\264\374\003\000\000\317\061\367\266\310\336\357\264\344J\f\000|\336ﴯ\363\002\000\254\020\n\000Info, \000\000Node"}
        ret = 3
#3  0x0002f594 in OpenZWave::LogImpl::Write(OpenZWave::LogLevel, unsigned char, char const*, std::__va_list) ()
No symbol table info available.
#4  0x0002eb3a in OpenZWave::Log::Write(OpenZWave::LogLevel, unsigned char, char const*, ...) ()
No symbol table info available.
#5  0x0007503a in OpenZWave::CentralScene::HandleMsg(unsigned char const*, unsigned int, unsigned int) ()
No symbol table info available.
#6  0x0005a6be in OpenZWave::Node::ApplicationCommandHandler(unsigned char const*, bool) ()
No symbol table info available.
#7  0x00055f32 in OpenZWave::Driver::ProcessMsg(unsigned char*) ()
No symbol table info available.
---Type <return> to continue, or q <return> to quit---
#8  0x00056b18 in OpenZWave::Driver::ReadMsg() ()
No symbol table info available.
#9  0x00057e0a in OpenZWave::Driver::DriverThreadProc(OpenZWave::Event*) ()
No symbol table info available.
#10 0xb4eff684 in ?? ()
No symbol table info available.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

Fishwaldo, 2015-10-18:
I need the OZW log file as well

sarakha63, 2015-10-18:
here it is ;)
```

<?xml version="1.0" encoding="utf-8" ?>
<poll homeid="0184d0d0" nodeid="1" sucnodeid="1" nodecount="2" cmode="SUC" save="1" noop="0">
        <log size="985" offset="68036">&#x0A;2015-10-18 18:16:58.905 Detail, Node023,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8&#x0A;2015-10-18 18:16:58.906 Detail, Node023,   ZW_SEND_DATA delivered to Z-Wave stack&#x0A;2015-10-18 18:16:58.921 Detail, Node023,   Received: 0x01, 0x05, 0x00, 0x13, 0x1b, 0x00, 0xf2&#x0A;2015-10-18 18:16:58.922 Detail, Node023,   ZW_SEND_DATA Request with callback ID 0x1b received (expected 0x1b)&#x0A;2015-10-18 18:16:58.923 Info, Node023, Request RTT 23 Average Request RTT 22&#x0A;2015-10-18 18:16:58.924 Info, Node023,   Node 23 has been marked as asleep&#x0A;2015-10-18 18:16:58.925 Detail,   Expected callbackId was received&#x0A;2015-10-18 18:16:58.926 Detail,   Expected reply was received&#x0A;2015-10-18 18:16:58.928 Detail,   Message transaction complete&#x0A;2015-10-18 18:16:58.929 Detail, &#x0A;2015-10-18 18:16:58.930 Detail, Node023, Removing current message&#x0A;2015-10-18 18:16:58.931 Detail, Node023, Notification: Notification - Node Asleep&#x0A;2015-10-18 18:16:58.932 Info, Notification: Notification home 0184d0d0 node 23 Sleep</log>
        <admin active="false" />
        <update />
</poll>
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=0
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=b4e03760
<?xml version="1.0" encoding="utf-8" ?>
<poll homeid="0184d0d0" nodeid="1" sucnodeid="1" nodecount="2" cmode="SUC" save="1" noop="0">
        <log size="0" offset="69021"></log>
        <admin active="false" />
        <update />
</poll>
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=0
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=b4e03760
<?xml version="1.0" encoding="utf-8" ?>
<poll homeid="0184d0d0" nodeid="1" sucnodeid="1" nodecount="2" cmode="SUC" save="1" noop="0">
        <log size="0" offset="69021"></log>
        <admin active="false" />
        <update />
</poll>
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=0
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=b4e03760
<?xml version="1.0" encoding="utf-8" ?>
<poll homeid="0184d0d0" nodeid="1" sucnodeid="1" nodecount="2" cmode="SUC" save="1" noop="0">
        <log size="0" offset="69021"></log>
        <admin active="false" />
        <update />
</poll>
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=0
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=b4e03760
<?xml version="1.0" encoding="utf-8" ?>
<poll homeid="0184d0d0" nodeid="1" sucnodeid="1" nodecount="2" cmode="SUC" save="1" noop="0">
        <log size="0" offset="69021"></log>
        <admin active="false" />
        <update />
</poll>
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=0
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=b4e03760
<?xml version="1.0" encoding="utf-8" ?>
<poll homeid="0184d0d0" nodeid="1" sucnodeid="1" nodecount="2" cmode="SUC" save="1" noop="0">
        <log size="0" offset="69021"></log>
        <admin active="false" />
        <update />
</poll>
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=0
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=b4e03760
<?xml version="1.0" encoding="utf-8" ?>
<poll homeid="0184d0d0" nodeid="1" sucnodeid="1" nodecount="2" cmode="SUC" save="1" noop="0">
        <log size="0" offset="69021"></log>
        <admin active="false" />
        <update />
</poll>
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=0
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=b4e03760
<?xml version="1.0" encoding="utf-8" ?>
<poll homeid="0184d0d0" nodeid="1" sucnodeid="1" nodecount="2" cmode="SUC" save="1" noop="0">
        <log size="0" offset="69021"></log>
        <admin active="false" />
        <update />
</poll>
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=0
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=b4e03760
<?xml version="1.0" encoding="utf-8" ?>
<poll homeid="0184d0d0" nodeid="1" sucnodeid="1" nodecount="2" cmode="SUC" save="1" noop="0">
        <log size="0" offset="69021"></log>
        <admin active="false" />
        <update />
</poll>
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=0
b5fff450: GET: "/poll.xml" conn=b6000468 size=0 *ptr=b4e03760
<?xml version="1.0" encoding="utf-8" ?>
<poll homeid="0184d0d0" nodeid="1" sucnodeid="1" nodecount="2" cmode="SUC" save="1" noop="0">
        <log size="0" offset="69021"></log>
        <admin active="false" />
        <update />
</poll>
2015-10-18 18:17:07.317 Detail, Node023,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x17, 0x05, 0x5b, 0x03, 0x12, 0x03, 0x04, 0xaf
2015-10-18 18:17:07.319 Detail,
Segmentation fault

```

Fishwaldo, 2015-10-19:
Should be fixed now.

By the way, that wasn't the OZW Log file. I'm talking about the actual OZW_Log.txt file, not the output of the ozwcp tool :)

closed 2015-10-19 by Fishwaldo

sarakha63, 2015-10-19:
You are right for the log file ;) my bad.

By the way the first tests seems successfull

Thanks again for your reactivity