OpenZWave/open-zwave #680: Qubino ZMNHIA2 config file update

meslow, 2015-10-26:
Made some config File updates for Qubino ZMNHIA2 config file.

Added missing parameter number : 4 - 5 - 10 - 11 - 12 - 60 and 61

https://groups.google.com/d/msg/openzwave/DDSakAm4N6w/sZjq7NVzAQAJ

Thank's

meslow, 2015-11-05:
I've update the config file but there's always an issue with COMMAND_CLASS_THERMOSTAT_SETPOINT.

I've try to moddify the value using OpenZWave Control Panel but after refresh le value is set back to 0.0 °C

Qubino comfimed the value can be SET and read using instance 1 class 0x43.

Fishwaldo, 2015-11-05:
I'd need a log file to see what's going on. Please remove any zwcfg_*.xml file when restarting your app

meslow, 2015-11-05:
I'm not very familiar with debbuging on Domoticz and OZW, whitch logs do you need and how to get them ?

I'll produce them a soon

meslow, 2015-11-05:
Here are some log from domoticz

[image]

nechry, 2015-11-05:
hello, 

I just complete a working config file in jeedom fork
https://github.com/jeedom/plugin-openzwave/commit/3a14ac8592141b0c4c67975ea29ec51785d301a2
it's work perfectly

we will merge very sone all pending config to openzwave

Fishwaldo, 2015-11-25:
Again, I'll wait for the Jeedom PR.

closed 2015-11-25 by Fishwaldo