OpenZWave/open-zwave #682: Unhandled Command Class 0x56

Stubbs, 2015-10-29:
Trying to work out why there are no zwave events appearing on my node-red flows, trying to see what raw data is being received and I'm seeing this from one of my devices, a Fibaro Motion Sensor (http://www.fibaro.com/en/the-fibaro-system/motion-sensor)

```
2015-10-29 21:09:31.581 Detail, Node006,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x06, 0x07, 0x56, 0x01, 0x30, 0x03, 0xff, 0xd1, 0xcb, 0x76
2015-10-29 21:09:31.582 Detail,
2015-10-29 21:09:31.582 Info, Node006, ApplicationCommandHandler - Unhandled Command Class 0x56
```

Can someone help me work out what the issue is?

Fishwaldo, 2015-10-30:
You need to ensure that the device reaches the “NodeQueriesComplete” stage. The message you post is usually a result of the device sleeping when OZW wakes up (and its in something like Probe stage) and we receiving a message form the device.

You can usually manually wake up the device by pressing buttons. Do that after starting OZW.

Fishwaldo, 2015-10-30:
(btw, please us the mailing list for general questions)

closed 2015-10-30 by Fishwaldo