OpenZWave/open-zwave #684: RefreshValue for sleeping node

ygoyal18, 2015-10-31:
Hello,

I am interfacing Fibaro motion sensor with OpenZwave. After configuring and associating the sensor I ran into the following scenario.

Setup: Node (Fibaro sensor) wake up duration = 60 seconds, Sensor_alarm cancellation delay = 5 seconds. 

Now if the sensor alarm triggers then the system tries to verify the reported change. But as the node is sleeping, the get command adds to the wakeup queue. Because the cancellation delay is lesser than the wakeup duration, sensor alarm value reverts back to the old value. Thus when the node wakes up that value changed is not reported and results in the loss of the event. 

Is there a way to cancel refresh value for only sleeping nodes?

Fishwaldo, 2015-11-01:
You have set "SetChangeVerified" on your Values? http://old.openzwave.com/dev/classOpenZWave_1_1Manager.html#ac26cc3c1b37cf068ac9eee23ef343c30 <http://old.openzwave.com/dev/classOpenZWave_1_1Manager.html#ac26cc3c1b37cf068ac9eee23ef343c30> 

I'd inly set that on devices that report incremental changes after a set (rather than the final value). Typically it's dimmers. You shouldn't set it on everything as its 99% of the time a waste of bandwidth. 

That being said, SetChangedVerified will never work "well" on battery devices.

Fishwaldo, 2015-11-01:
Please use the mailing list for further questions.

closed 2015-11-01 by Fishwaldo