OpenZWave/open-zwave #685: Qubino ZMNHLA2.xml config File

meslow, 2015-11-02:
I've just create config Faile for ZMNHLA2 Qubino Flush PWM thermostat.

Sorry a couldn't test it.

https://groups.google.com/forum/#!topic/openzwave/7JsvzTreQGw

Regards,

nechry, 2015-11-10:
hello, I found little mistake in your config file (association groups, This thermostat's setpoint descriptions are 0 based, not 1), 
we (jeedom) will propose really soon a PR to ozw  

see ZMNHLAx
https://github.com/jeedom/plugin-openzwave/blob/master/ressources/openzwave/config/qubino/ZMNHLAx.xml
I choose x at the end to covert all freq region
ZMNHLA2 868,4 MHz
ZMNHLA3 921,4 MHz
ZMNHLA4 908,4 MHz
ZMNHLA5 869,0 MHz
ZMNHLA6 916,0 MHz

Fishwaldo, 2015-11-25:
I'll wait for the Jeedom PR instead.

closed 2015-11-25 by Fishwaldo