OpenZWave/open-zwave #687: pkg-config stuff is missing for libopenzwave1.3-dev package

pastukhov, 2015-11-03:
Seems like libopenzwave.pc generation is wrong and missing in libopenzwave1.3-dev package creation spec.

Fishwaldo, 2015-11-04:
Wrong? In what way. Can you provide a bit more detail than a one liner please?

pastukhov, 2015-11-05:
I'm trying to build packages for Debian jessie directly on my Banana Pi:
```dpkg-buildpackage -b```
* libopenzwave.pc file is missing from libopenzwave1.3-dev_0.9_armhf.deb.
I found it in ./debian/tmp/usr/local/lib/arm-linux-gnueabihf/pkgconfig/libopenzwave.pc:
```
prefix=/usr
exec_prefix=/usr/bin
libdir=/usr/lib/arm-linux-gnueabihf/
includedir=/usr/include/openzwave/
sysconfdir=/usr/etc/openzwave/
gitversion=0.9.-1
docdir=/usr/share/doc/openzwave-0.9.0/

Name: libopenzwave
Description: A Open Source implementation of the ZWave Serial API
URL: http://www.openzwave.org/
Version: 0.9
Libs: -L${libdir} -lopenzwave
Cflags: -I${includedir} 
```
* So it has been generated in wrong place and having wrong version number.

Fishwaldo, 2015-11-16:
Could you please post the output of "pkg-config --variable pc_path pkg-config"

Fishwaldo, 2015-11-24:
Pending on a fix from PR #689

jjhuff, 2017-10-06:
Curious if anyone has a solution for this. I'm using the Debian stretch package, and it's missing libopenzwave.pc (https://packages.debian.org/stretch/amd64/libopenzwave1.5-dev/filelist)

Thanks!

closed 2019-05-13 by Fishwaldo