OpenZWave/open-zwave #688: Qubino ZMNHDD1 Z-Wave Plus flush dimmer config file

bbqkees, 2015-11-03:
I made the configuration file for the Qubino ZMNHDD1 Z-Wave Plus flush dimmer.
My colleague tested it and it works just fine.

Addition for the manufacturer_specific.xml:
[code]<Product type="0001" id="0051" name="ZMNHDD1 Z-Wave Plus Flush Dimmer" config="qubino/ZMNHDD1.xml"/>[/code]

Files are here:
https://groups.google.com/forum/#!topic/openzwave/7h1TU-PiTHU

StefanPuntNL, 2015-11-10:
Please double check the config file before implementing it. Some things I noticed:

- Parameter 1: default value is set to 1, should be 0.
- Parameter 10: help text says default value is 0, but should say 255.
- Parameter 100 and 101: default value set to 0, but according to manual default value is 1

bbqkees, 2015-11-10:
After I checked it again, I indeed found several errors.
Included the modified version. Should be complete now.

StefanPuntNL, 2015-11-10:
Nice! Thanks for the effort :+1:

bbqkees, 2015-11-10:
One question:
The manual lists several parameters as '2 Byte'.
Should I use Byte or Short here? Other Qubino config files use Byte every time.

StefanPuntNL, 2015-11-10:
Value Type Byte  = 8-bit (1 byte) unsigned value (−128 to 127)
Value Type Short = 16-bit (2 bytes) signed value (−32768 to 32767)

It is possible that others have created invalid config files. But 2 byte values should be of type short.
 .

bbqkees, 2015-11-10:
Then let me change the config once again.

StefanPuntNL, 2015-11-10:
Please see my updated response. I corrected the min max values of the unsigned 8-bit value.

bbqkees, 2015-11-10:
Funny thing is f.i. parameter 66 has a min-max of 1-255 but they list it as: "data type is 2 Byte DEC".
So you would expect 255 to fit in one byte.

StefanPuntNL, 2015-11-10:
I think that somewhere there is a signed/unsigned value conversion. Signed 8-bit value is -128 to 127.

bbqkees, 2015-11-10:
Yes i think you are correct.

Below the latest config with some changes from type byte to short:
https://groups.google.com/forum/#!topic/openzwave/7h1TU-PiTHU

nechry, 2015-11-10:
Only Associations group 1 must be set auto="true". 
All other must set to false or simply remove the attribute to more readability.

Also I'm not sure the  Map endpoints to instances must be set. (but I don't have the device to validate, I hope you set for a reason)

bbqkees, 2015-11-11:
I used another existing Qubino config file as an example in the assumption they were all correct.
Apparently not :-)

Endpoint mapping is empty, I am not sure what you mean.

I had a discussion with Qubino and in Q1 2016 they planned to provide all the config files themselves.
But of course until then we need to take care of them ourselves.

New file:
https://groups.google.com/forum/#!topic/openzwave/7h1TU-PiTHU

nechry, 2015-11-11:
I'm talk about this part:

<CommandClass id="96" mapping="endpoints"/>

It's a good news if qubino will provide there file. 
we (jeedom) will merge soon other qubino configuration tested with devices
https://github.com/jeedom/plugin-openzwave/tree/master/ressources/openzwave/config/qubino
I don't have a ZMNHDD1  to test this config do you have one?

we "normally" only propose PR for configuration tested, it's why we don't merge quickly to openzwave in general there config a fully tested, but mistake can also happen.

bbqkees, 2015-11-11:
So what part exactly?

I have a ZMNHDD1 and a colleague of mine too but we are using Domoticz als home automation software.

nechry, 2015-11-11:
sorry I fix the markdown to show xml
but if you have tested this configuration is good.

you also remove CC basic any reason?

bbqkees, 2015-11-11:
I copied the endpoint mapping from another Qubino config file so I did not change anything there.

Did not change any CC on purpose.

The current config is working fine for my colleague.

nechry, 2015-11-11:
for me there attributes must be set for a reason, each qubino have different behavior.

so if they work go for it, I don't have the device.

Fishwaldo, 2015-11-11:
Do you have a contact at Qubino I can chat too? Last time I tried there support I was told to refer my questions to "the people that manufacture the gateway I'm using" (so they obviously didn't know anything about OZW)

https://groups.google.com/forum/#!topic/openzwave/7h1TU-PiTHU

Fishwaldo, 2015-11-11:
I'm happy to merge "works for me" configs files but I'm on vacation at the moment and I've had about 30 seconds in front of a laptop in the last week. 

When I'm back I'll merge the files. If anybody else can test before then it's even better!

nechry, 2015-11-11:
hi Justin we will got a ZMNHDD1 until your end of Holiday, so we will test the last proposed configuration and fix it if we detect something.

for a qubino contact, from my side I actually don't have any contact. But I have started a little chat with qubino support about ozw and jeedom as a gateway, so wait and see.

enjoy your holiday, you deserve it! 
Cheers

bbqkees, 2015-11-11:
Well I send an email to info@qubino.com. So no specific contact.
(Their respone is quite flaky).

I asked them to put some effort in OpenZWave beause as they do not make a controller they owe a lot of sales to OpenZWave. Delivering 100% perfect configuration files ensures great operabillity and letting the public do all the work instead will not give the best results. 

So their response was they were already looking into it and expecting to deliver config for new (and older devices) in Q1 2016.

Fishwaldo, 2015-11-15:
Could you forward that email to me so I can follow up with them direct? I'd like to offer whatever assistance they need.

bbqkees, 2015-11-16:
Yes i can but I do not want to publish his email address plain on the internet.

Fishwaldo, 2015-11-16:
Justin at dynam dot ac

Fishwaldo, 2015-11-25:
Commited. Thanks!

closed 2015-11-25 by Fishwaldo