OpenZWave/open-zwave #691: Everspring HAC01

FafaDomo, 2015-11-08:
<?xml version="1.0" encoding="utf-8"?>

<Product xmlns='http://code.google.com/p/open-zwave/'>
  
  <CommandClass id="112">
    <Value type="byte" genre="config" index="1" label="Basic Set Level" min="0" max="99" value="1">
      <Help>Basic Set Level
     0 -> Off
     1 - 99 -> On(binary switch device); dim level (multilevel switch device)
     </Help>
    </Value>
   
    <Value type="byte" genre="config" index="2" label="Amount of delay" min="0" max="127" value="0">
      <Help>determine the delay time setting to send the Basic Set Off command when manual switch opens
     0 -> immediately Off
     1 - 127 -> delay 1 ~ 127 second to Off
     </Help>
    </Value>
  </CommandClass>
 
  <CommandClass id="32" setasreport="true" />
  <CommandClass id="96" mapping="endpoints" />
 
  <CommandClass id="133">
    <Associations num_groups="1">
      <Group index="1" max_associations="5" label="Group 1" auto="true"/>
    </Associations>
  </CommandClass>

</Product>

nechry, 2015-11-08:
it's a missing device configuration? We need product and type and id. Check wiki for adding device
https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

FafaDomo, 2015-11-08:
Yes, it is ! Sorry I'm a newbie, it's my first zwave conf file...
<Manufacturer id="0060" name="Everspring"> (already in database)
Product Type ID: 	0x0010
Product ID: 	0x0001
Product Name: 	In-wall remote module HAC01
Brand Name: 	Everspring
Product Code: 	EVR_HAC01

Fishwaldo, 2015-11-16:
its all mangled. Can you please post the config file to the mailing list (or open a Pull Request).

Fishwaldo, 2015-11-25:
Ping. Please repost the changes to the mailing list, otherwise Ill close this.

Fishwaldo, 2015-11-30:
Closing as no response

openzwave-dev-watcher, 2016-01-11:
Hi, may i reopen this thread ? I found that device configuration on the 
domoticz french website i saved it as HAC01.xml in Everspring dir.
cat manufacturer_specific.xml | grep hac

 <Product type="0010" id="0001" name="HAC01 In-wall Remote Module" config=
"everspring/hac01.xml"/>

cat everspring/hac01.xml 

<?xml version="1.0" encoding="utf-8"?>

<Product xmlns='http://code.google.com/p/open-zwave/'>
  
  <CommandClass id="112">
    <Value type="byte" genre="config" index="1" label="Basic Set Level" min=
"0" max="99" value="1">
      <Help>Basic Set Level
     0 -> Off
     1 - 99 -> On(binary switch device); dim level (multilevel switch 
device)
     </Help>
    </Value>
   
    <Value type="byte" genre="config" index="2" label="Amount of delay" min=
"0" max="127" value="0">
      <Help>determine the delay time setting to send the Basic Set Off 
command when manual switch opens
     0 -> immediately Off
     1 - 127 -> delay 1 ~ 127 second to Off
     </Help>
    </Value>
  </CommandClass>
 
  <CommandClass id="32" setasreport="true" />
  <CommandClass id="96" mapping="endpoints" />
 
  <CommandClass id="133">
    <Associations num_groups="1">
      <Group index="1" max_associations="5" label="Group 1" auto="true"/>
    </Associations>
  </CommandClass>

</Product>

openzwave-dev-watcher, 2016-01-18:
> You received this message because you are subscribed to the Google
> Groups "OpenZWave" group.
> To unsubscribe from this group and stop receiving emails from it,
> send an email to openzwave+unsubscribe@googlegroups.com.
> For more options, visit https://groups.google.com/d/optout.

Please open a new issue on Github.

Fishwaldo, 2016-02-01:
thats fine.

nechry, 2016-06-08:
already existe:
https://github.com/OpenZWave/open-zwave/blob/master/config/manufacturer_specific.xml#L293

closed 2016-06-08 by nechry