OpenZWave/open-zwave #692: Error in manufacturer_specific.xml

gsolem, 2015-11-09:
I'm sending this as an issue as I am not able to pull. There is an error for TKB TZ68 in manufacturer_specific.xml, it should be:

<Manufacturer id="0118" name="Wenzhou TKB Control System">
<Product type="0101" id="0103" name="TZ68 On/Off Switch Socket" config="wenzhou/tz68.xml"/>
</Manufacturer>

The corresponding xml-file for the unit should be like this:

<?xml version="1.0" encoding="utf-8"?>
<Product xmlns='http://code.google.com/p/open-zwave/'>
	
        <CommandClass id="112">
                <Value type="byte" genre="config" instance="1" index="1" label="LED behavior" min="0" max="1" value="0">
                        <Help>
                                0 -> night light - LED ON when the load attached is turned OFF (default).
                                1 -> on indicator - LED ON when the load attached is turned ON.
                        </Help>
                </Value>
        </CommandClass>
</Product>

nechry, 2015-11-10:
you talk about TZ66D Dual Wall Switch ?
		<Product type="0102" id="1020" name="TZ66D Dual Wall Switch" config="wenzhou/tz66d.xml"/>
parameter 3?
<Value type="byte" genre="config" instance="1" index="3" label="Night light" min="0" max="1" value="0">
		  <Help>The LED on the TZ66D will by default, turn ON when the load attached is turned OFF. To make the LED turn ON when the load attached is turned ON instead, set parameter to a value of 1.</Help>
		</Value>

Night light
Parameter No: 3
Length: 1 Byte
Valid values = 0 or 1 (default value is 0)
The LED on the TZ66-D will by default, turn ON when the load attached is turned OFF.
To make the LED turn ON when the load attached is turned ON instead, set parameter 3 to a value of 1.

nechry, 2015-11-10:
you mean:
LED- Behavior (Parameter Number 1, Parameter Size 1) defines the behavior of the blue LED
Value Description
0 blue indicates off (Default)
1 blue indicates on

http://www.vesternet.com/downloads/dl/file/id/59/product/899/z_wave_tkb_on_off_switch_socket_uk_manual.pdf

caederus-openzwave, 2015-11-10:
Looks like the XML got mangled in the OP - I presume the intention was to quote this, which I added to [Domoticz](https://www.domoticz.com/forum/viewtopic.php?f=24&t=2002) in March:
```xml
<?xml version="1.0" encoding="utf-8"?>

<Product xmlns='http://code.google.com/p/open-zwave/'>
        <!-- Configuration -->
        <CommandClass id="112">
                <Value type="byte" genre="config" instance="1" index="1" label="LED behavior" min="0" max="1" value="0">
                        <Help>
                                0 -> night light - LED ON when the load attached is turned OFF (default).
                                1 -> on indicator - LED ON when the load attached is turned ON.
                        </Help>
                </Value>
        </CommandClass>
</Product>
```

I'm not sure where I got the "night light" text originally, but it's more like the existing openzwave help for an identical option in tz65d and tz66d:
```xml
<Value type="byte" genre="config" instance="1" index="3" label="Night light" min="0" max="1" value="0">
                  <Help>The LED on the TZ65D will by default, turn ON when the load attached is turned OFF. To make the LED turn ON when the load attached is turned ON instead, set parameter to a value of 1.</Help>
</Value>
```
The TZ67 (config/wenzhou/tz67.xml) already has the brief text nechry mentions; it may be endorsed by the manufacturer (or the very helpful Vesternet?) but I think it is rather too cryptic - it's only understandable once you know what it means!

nechry, 2015-11-10:
Only Associations group 1 must be set auto="true". 
All other must set to false or simply remove the attribute to more readability.

Also I'm not sure the  Map endpoints to instances must be set. (but I don't have the device to validate, I hope you set for a reason)

gsolem, 2015-11-12:
That is right, seems like the xml got truncated.

Fishwaldo, 2015-11-16:
I'm not clear on what needs to change. Can you clarify?

gsolem, 2015-11-16:
The LED on the TZ68 switch cannot be configured today. In manufacturer_specific.xml for manufacturer ID 0118 the following needs to be added:

```<Product type="0101" id="0103" name="TZ68 On/Off Switch Socket" config="wenzhou/tz68.xml"/>```

Then the following included as tz68.xml in the wenzhou folder:
```
<?xml version="1.0" encoding="utf-8"?>

<Product xmlns='http://code.google.com/p/open-zwave/'>

	<!-- Configuration -->
        <CommandClass id="112">
                <Value type="byte" genre="config" instance="1" index="1" label="LED behavior" min="0" max="1" value="0">
                        <Help>
                                0 -> night light - LED ON when the load attached is turned OFF (default).
                                1 -> on indicator - LED ON when the load attached is turned ON.
                        </Help>
                </Value>
        </CommandClass>
</Product>
```

closed 2015-11-24 by Fishwaldo

gsolem, 2015-11-30:
Please remove this line from manufacturer_specific.xml as it overrides the fix included in this issue:

`<Product type="0101" id="0103" name="TZ68E On/Off Switch Socket"/>`