OpenZWave/open-zwave #695: prepend value labels with a instance name

Fishwaldo, 2015-11-16:
Allow us to specify instance names in the config files, and then prepend value labels with the instance name, so you can have things like:
"Outlet Level" and "LED Level" on things like the Aeotec Smart Dimmer 6 (instance 1 is the outlet, instance 2 is the LED ring)

nechry, 2015-12-01:
nice feature I recently include a Aeotec Smart Dimmer 6 and yes it's cant be nice to have this information. Is also the reason we see on Aeotec Smart Switch 6 a SwitchMultiLevel CC, at beginning I'm thinking about a wrong interview but it's also the LED ring.

vincemic, 2019-01-27:
I'd like to take on the challenge of adding the enhancement. I would like to possibly add a new XML element under Product in device_configuration.xsd. Not sure what it should be called. It would have an id, name, and a description of the "subdevice" or "endpoint". Then each value could have a subdeviceid reference. This would be helpful for power strips that have multiple binary switches and meters.  In the case of the aeotec power strip, it would have 5 subdevices. The first subdevice name would be "Main Power", the others would be "Socket 1", "Socket 2". 

Figuring out the XSD is the first part. Then how would you want the API to change to retrieve the additional data. The simple way may be to augment the value to bring back the whole subdevice object if it has a reference. This would make the change backwards compatible since it could be ignored.

Let me know if you already had a solution in mind.

vincemic, 2019-01-28:
Ok, makes sense. There is one problem though.  I would like to associate more than one command instance to a "subdevice". For the Aeotech dimmer socket, there is a night light that has at least two command class instances associated to it. Multilevel Switch and Color. Both would affect the night light only and if I change the name "Night Light" to something else, I would want both command instances to share the change. To complicate things, The multilevel switch for the night light is instance 2 and the color command for the night light is instance 1. 

If we had an object that was "subdeivce" or some better name, we could name it "Night Light" and assign it one or more command instances.

When presenting these objects in a UI or in voice commands, I would have the application layer show two virtual devices showing as lights or light switches. One would be  "Aeotec Dimmer" the other would be "Night Light".  You can rename either independently.

Fishwaldo, 2019-01-29:
I think you can still do that with two `<instance>` blocks. Nothing says we can't have duplicate names for each instance.
```xml
<Instance id="2" label="Night Light">
    <label lang="fr">veilleuse</lang>
</instance>
<Instance id="3" label="Night Light">
    <label lang="fr">veilleuse</lang>
</instance>
```

Fishwaldo, 2019-01-29:
I think its ok to save them...

vincemic, 2019-01-30:
If the class instance's ids where mapped correctly by endpoint then the solution for instance labels would work perfectly.

vincemic, 2019-01-30:
Adding log with additional log around endpoint and instance mapping
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/2814232/OZW_Log.txt)

vincemic, 2019-01-31:
After looking through the code, I see that it was specifically designed to allow the id of an instance to stray from the associated endpoint id. I think that changing the way instances work will possibly break bindings end users have already setup.

I would like to offer this solution. I would like to expose endpoints as a first class citizen. In configuration you would see the following

<endpoint id="1" label="Socket">
    <label lang="fr">prise</lang>
</endpoint>
<endpoint id="2" label="Night Light">
    <label lang="fr">veilleuse</lang>
</endpoint>

But in order to present things correctly we will need to expose endpoints through the manager. First I would like to augment the ValueID to include endpoint id. Endpoint id shouldn't change the m_id packing or the m_id2 packing. Then add the methods for endpoint.

Manager::GetEndpointName(uint32 const _homeId. uint8 const _nodeId, uint8 const _endpointId, int32 _pos = 1);
Manager::SetEndpointName(uint32 const _homeId. uint8 const _nodeId, uint8 const _endpointId,  string const& name, int32 _pos = 1);

This would allow applications that are endpoint aware to associate ValueID to endpoints and give those endpoints a descriptive name.

Fishwaldo, 2019-02-01:
> 
> 
> So this may be a bug, not sure. OZW knows with the MultiChannelEndPointReport that the endpoints are not the same for the Aeotec Dimmer 6. Not sure if instances and endpoints are meant to be the same thing. Is OZW stacking command instances and giving them an index in arrival order. If instance id was the same as endpoint id then it might be possible for the Color command class to have a instance 2 but no instance 1. Still learning the code and terminology.

instances and endpoints are essentially representing the same thing, except, instance 1 is always the root node, and endpoint 1 is the first node under the root. 

The instances are "stored" against valueID's and not command classes. So one CommandClass on a Node can work with multiple instances/endpoints. 

So in your case, you would have a (few) ValueID that are of the User Genre for the ColorCC on Instance 2, but not on Instance 1 (if you do have instance 1, thats a bug).

Fishwaldo, 2019-02-01:
> 
> 
> Adding log with additional log around endpoint and instance mapping. Watch the color command class
> [OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/2814232/OZW_Log.txt)

Just a FYI - I'd do your development with no security. This is due to WIP on #1639 which is related to MultiInstance issues.

Fishwaldo, 2019-02-01:
> 
> 
> After looking through the code, I see that it was specifically designed to allow the id of an instance to stray from the associated endpoint id. I think that changing the way instances work will possibly break bindings end users have already setup.
> 
> I would like to offer this solution. I would like to expose endpoints as a first class citizen. In configuration you would see the following
> 
> ```
> <endpoint id="1" label="Socket">
>     <label lang="fr">prise</lang>
> </endpoint>
> <endpoint id="2" label="Night Light">
>     <label lang="fr">veilleuse</lang>
> </endpoint>
> ```
> 
> But in order to present things correctly we will need to expose endpoints through the manager. First I would like to augment the ValueID to include endpoint id. Endpoint id shouldn't change the m_id packing or the m_id2 packing. Then add the methods for endpoint.
> 
> ```c
> Manager::GetEndpointName(uint32 const _homeId. uint8 const _nodeId, uint8 const _endpointId, int32 _pos = 1);
> Manager::SetEndpointName(uint32 const _homeId. uint8 const _nodeId, uint8 const _endpointId,  string const& name, int32 _pos = 1);
> ```
> 
> This would allow applications that are endpoint aware to associate ValueID to endpoints and give those endpoints a descriptive name.

How do they stray? as mentioned above, instances are the same as endpoints, its just 0 based versus 1 based counting thats makes them different. 

as Instances = Endpoints, I don't see how this helps you. Your still going to have dynamic endpoints with different CC's. 

I'm lost as far as what you are trying to show in the logs. Can you expand on what you are seeing?

stiansoevik, 2019-04-08:
Regarding bit masks, I read through an app note while working on another issue, which briefly mentioned the rationale for allowing bitmasks. It makes sense to be able to e.g. control all switches simultaneously.

https://www.silabs.com/documents/login/application-notes/APL12955-Z-Wave-Multi-Channel-Basics.pdf

Fishwaldo, 2019-04-24:
So I've done this, and hopefully in a way that addresses your challenge with the endpoint mappings. 

If you Specify Instance Labels for the MultiInstance CC (in the config file) they are applied to all Instances/CommandClasses (ie, global). But you may also add Instance Labels to individual CommandClasses as well, to give them different names that the Global Ones. 

If No Instances Labels are specified in the Config File, it defaults to "Instance x"

ValueLabels are also prepended with the Instance Label so you can easily identify what Control relates to what instance without having to display the actual instance Number

This may be disabled by Setting "IncludeInstanceLabel" as false in the Options.xml or class. 

A example (Global) instance config file: https://github.com/OpenZWave/open-zwave/commit/a64edfb753d5cd815583045078c37352c130b478#diff-7693ba9a9c6991d55e25c52dbf0df288

(Just add another commandclass block if you want different instance labels for different commandclasses).

commit a64edfb7 (2019-04-24): Implement Labels for Instances - Allows a friendly name describing what each instance does. Fixes #695
closed 2019-04-24 by Fishwaldo