OpenZWave/open-zwave #698: config/vision/zp3102.xml update

Erik-NA, 2015-11-19:
Added parameters for Vision ZP3102 IN|MY|EU|RU|US|IL|KR|HK|JO|BR PIR Motion Sensor.
Celcius/Fahrenheit
Sensitivity
Temperature adjustment

[Uploading zp3102.xml.txt…]()

<?xml version="1.0" encoding="utf-8"?>

<Product xmlns='http://code.google.com/p/open-zwave/'>
        
        
        <CommandClass id="112">
				<Value type="byte" genre="config" index="1" label="Re-Trigger Waiting Time" units="minutes" min="1" max="255" value="3">
					<Help>Time until an associated device will turn off if no new movement is detected</Help>
                </Value>
                <Value type="list" genre="config" index="2" label="Celcius/Fahrenheit" size="1" min="0" max="1" value="0">
					<Help>Report temperature in Celcius or Fahrenheit</Help>
					<Item label="Celcius" value="0" />
					<Item label="Fahrenheit" value="1" />
                </Value>
                <Value type="byte" genre="config" index="3" label="Sensitivity" units="" min="1" max="7" value="4">
					<Help>Sensitivity 1-7 (1 = most sensitive, 7 = most insensitive. 4 is default)</Help>
                </Value>
                <Value type="byte" genre="config" index="4" label="Temperature adjustment" units="Degrees Celsius" min="0" max="255" value="0">
					<Help>-10 to -1, 0 to 10 (Signed decimal: 246 to 255, 0 to 10)</Help>
                </Value>
        </CommandClass>

        
        <CommandClass id="32" setasreport="true"/>

        
        <CommandClass id="133">
                <Associations num_groups="1">
                        <Group index="1" max_associations="5" label="Basic on/off group for detecting movements" auto="true"/>
                </Associations>
        </CommandClass>

</Product>

closed 2015-11-24 by Fishwaldo

Erik-NA, 2015-12-11:
I cannot see this in the code tree. Is it really merged?

Erik-NA, 2015-12-15:
As I can see, the file content of vision/zp3102.xml is not modified according to above?

Fishwaldo, 2015-12-21:
Sorry, I added the file as the wrong name.