OpenZWave/open-zwave #703: Updated Configuration file for BeNext Doorsensor

gizmocuz, 2015-11-28:
Attatched a new configuration file for the benext doorsensor

- added missing configuration options, added default values in help
- ignoring get-alarm class (113)
- auto association in group one (else nothing is received) 

[DoorSensor.xml.txt](https://github.com/OpenZWave/open-zwave/files/46292/DoorSensor.xml.txt)

Fishwaldo, 2015-11-30:
Group one is always set to Auto automatically, so no need to put that it, otherwise its ok.

closed 2015-11-30 by Fishwaldo

gizmocuz, 2015-11-30:
Strangly... it was not... i had to always add node 1 to group 1
with this, it was added automatically
... is there a configuration setting i need to enable to automatically assign node 1 to group 1 ?

i am using now

OpenZWave::Options::Get()->AddOptionBool("Associate", true);

gizmocuz, 2015-11-30:
I had another user, he had 6 nodes, none had node1 in group 1

Fishwaldo, 2015-11-30:
https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/Group.cpp#L67

if it doesn't work, need Logs.

gizmocuz, 2015-11-30:
Could it be, that in the original config file the group 1 index was set to auto="false"

<Group index="1" max_associations="5" label="Group 1: if the internal door contact (magnet) is triggered it sends a Z-Wave frame to every node in this group." auto="false"/>

i removed the 'auto' part

included the node... inclusion successful, but nothing is received... node 1 not in controller

please see attached the log, looks like we need auto=true

i included the node again, please see OZW_Log_benext_2.txt, now it works, needed the auto=true

[OZW_Log_benext2.txt](https://github.com/OpenZWave/open-zwave/files/47588/OZW_Log_benext2.txt)

[OZW_Log_benext.txt](https://github.com/OpenZWave/open-zwave/files/47586/OZW_Log_benext.txt)

Fishwaldo, 2015-12-01:
auto="false" will disable that automatic associations on Group 1

I'm looking through the code right now, but stumped. Will investigate some more