OpenZWave/open-zwave #704: Auto Association on Group 1 isn't working

Fishwaldo, 2015-12-01:
Per Issue #703

nechry, 2015-12-01:
With old fibaro devices the Lifeline/Report is on group 3, so it's not good to assume the group 1 is for controller, in my point of view.

Fishwaldo, 2015-12-02:
Every other device I've seen has the LifeLine, or at least a Basic Set on Group 1. Changing the default, for a company that we know "plays loose and fast" with the protocol doesn't seem right to me. :)

openzwave-dev-watcher, 2015-12-02:
i agree with this, group 1 should contain the controller by default

And users should read the manual, so in 0.5% of this case it should not be 
in group 1
Also some hardware sends updates to group 3, and after reading the manual a 
user can decide if it should add the controller in group 3

The problem is, almost nobody is reading the manual, and for these users 
its 99.5% the case that the controller should be added to group 1

Currently it is raining complaints.... this node ain't working good... and 
in almost all the cases the controller is not in group 1

nechry, 2015-12-02:
There is a way, if for some reason we don't have the controller auto associate to group 1 ? I meen in device config the auto to false will not overrided? But yes many time is a issue, we wrote a zwave heal status page and add a warning in case the controler is not present at least in one group. But most case is ok with good device config file.

Fishwaldo, 2015-12-03:
If Auto is false on Group 1 it should not associate

Fishwaldo, 2015-12-11:
Ok. I figured out what was going on. If a device had a config file, then the autoassoicate wouldn't work. Only devices that didn't have a config file would auto associate. 

I've fixed it so regardless all devices will default to AutoAssociate on Group 1, and it can be overriden by setting auto=false in the Association config.

commit f32e84b7 (2015-12-11): Fixed #704 - AutoAssoicate should work regardless of Config file present or not
closed 2015-12-11 by Fishwaldo