OpenZWave/open-zwave #714: Z-Wave.Me ZME_KFOB-C 4 Button Key Fob

pbda, 2015-12-16:
The new Zwave.me KFOB-C 4 buttons keyfob is very similar to KFOB-S model and can use same config file `zwave.me/ZME_KFOB-S.xml` but has a new `id` "0103".

    # diff manufacturer_specific.xml manufacturer_specific.xml.sv
    769,770c769
    < 		<Product type="0100" id="0102" name="ZME_KFOB-S 4 Button Key Fob" config="zwave.me/ZME_KFOB-S.xml" />
    < 		<Product type="0100" id="0103" name="ZME_KFOB-C 4 Button Key Fob" config="zwave.me/ZME_KFOB-S.xml" />
    ---
    > 		<Product type="0100" id="0102" name="ZME_KFOB-S 4 Button Key Fob"/>

Fishwaldo, 2015-12-21:
We dont' ahve a ZME_KFOB-S.xml file. Can you please attach a copy?

pbda, 2015-12-21:
I get it here:
https://github.com/jeedom/plugin-openzwave/blob/master/ressources/openzwave/config/zwave.me/ZME_KFOB-S.xml
I didn't notice it was not OpenZWave Git :-o

closed 2015-12-22 by Fishwaldo