OpenZWave/open-zwave #716: Aeotec Multisensor6 calibration setting

michapr, 2015-12-22:
Calibration settings of Aeotec Multisensor6 does not work.
If you will set Temperature calibration or Humidity calibration - value will be ignored.
What is interesting - the Luminance Calibration and Ultraviolet Calibration are working fine.

In Result of the changes you see in log:
2015-12-22 12:19:20.223 Info, Node073, Received Configuration report: Parameter=203, Value=50
2015-12-22 12:21:28.504 Info, Node073, Received Configuration report: Parameter=204, Value=5
2015-12-22 12:21:55.214 Info, Node073, Received Configuration report: Parameter=201, Value=0
2015-12-22 12:21:55.465 Info, Node073, Received Configuration report: Parameter=202, Value=0

So only parameters 201/202 does not work - have no idea what this can be...

Attached a working log (Luminance) and failed log (Temp.)
[Luminance_calibration.zip](https://github.com/OpenZWave/open-zwave/files/69632/Luminance_calibration.zip)
[Temp_calibration.zip](https://github.com/OpenZWave/open-zwave/files/69633/Temp_calibration.zip)

Michael

michapr, 2015-12-22:
Found out the problem!
Changing the config from "short" to "byte" - and all will be ok.
Value type="byte" index="201" genre="config" label="Temperature Calibration" value="0"
....
Value type="byte" index="202" genre="config" label="Humidity Calibration" value="0"

Please check - and if it is ok, change the multisensor6.xml please ;)

Michael

nechry, 2015-12-22:
yes see my version, not yet merged
https://github.com/jeedom/plugin-openzwave/blob/master/ressources/openzwave/config/aeotec/multisensor6.xml
I fix all default values and other small thing. Aeon-labs have updated the ing sheet all is in phase with latest

michapr, 2015-12-22:
So... should be merged... When it could be done?
Then users must not edit the config xml more ;)

Thanks!

nechry, 2015-12-22:
Yes we have many PR in pending. But the merge in on hold unitl we update the zwave plugin for jeedom to match auto controller association changes

Fishwaldo, 2015-12-24:
Your going to have many more of these "differences" soon Nechry, once I start work on the next version, you can expect the config file format to be updated again.... 

Commited the fix to repo as well.

closed 2015-12-24 by Fishwaldo