OpenZWave/open-zwave #719: Wenzhou TKB TSM02 4 in 1 Door/Window Sensor

ertanden, 2015-12-25:
I have a "Wenzhou TKB TSM02 4 in 1 Door/Window sensor", but with latest Open-ZWave it's being identified as a "TSP01 3 in 1 PIR Motion Sensor", which causes issues with door/window and PIR sensors.

TSM02 is exactly the same as "PST02-A 4 in 1 Multi-Sensor", so I made the following change and the sensor works as expected:

In file`config/manufacturer_specific.xml`, change line:

    <Product type="0002" id="0002" name="TSP01 3 in 1 PIR Motion Sensor" config="wenzhou/tsp01.xml" />

to

    <Product type="0002" id="0002" name="TSP01 3 in 1 PIR Motion Sensor" config="philio/pst02.xml" />

This of course will cause TSP01 sensors not work as expected. I'm pretty new to Open-ZWave so I couldn't figure out how to add a device instead of replace, specifically what should be written to the `type` and `id` values...

Fishwaldo, 2015-12-28:
The manufacturer has reused Type and ID codes here. Nothing OZW can do about it. You should log a case with Wenzhou asking them which is which.

closed 2015-12-28 by Fishwaldo

ertanden, 2015-12-29:
Well, ok they look like they have the same id values. But, maybe we can have a solution for both to be supported. In order to support both TSM02 and TSP01, would it be sufficient to update `config/wenzhou/tsp01.xml` with content from `config/philio/pst02.xml`?

Maybe that will make the code work both for TSM02 and TSP01 since TSM02 is a superset of TSP01, it just has an extra door/window sensor in addition to PIR, illumination and temperature that's available in TSP01. All config values are the same for both TSM02 and TSP01 as far as I can see.

TSP01 manual:
https://www.intellihome.be/en/productattachments/index/download?id=1101

TSP02 manual:
https://www.intellihome.be/en/productattachments/index/download?id=1100

Fishwaldo, 2016-01-02:
OZW unfortuantly doesn't support duplicate Product/Type ID's. 
But if the configs are the same, at most, you just end up with the
wrong product name....

ertanden, 2016-01-03:
Configs are the same but the reported values are not, so it's not just an issue of a wrong name. 

Anyways, I don't know how these XMLs work so can't really propose a solution, but my previous question looks to be ignored.

TSM02 is a superset of TSP01, it just has an extra value reporting Door/Window contact. Would it be possible to support both TSM02 and TSP01 together, even if they have the same IDs? Since Philio PST02 is same as TSM02 I thought maybe replacing content of `config/wenzhou/tsp01.xml` with content from `config/philio/pst02.xml` would work?

Fishwaldo, 2016-01-04:
As I mentioned, we don’t support duplicate ID’s. It won’t work.

ertanden, 2016-01-06:
Well, I'm not asking for support of duplicate IDs. I'm asking to update a config to support both devices.

Basically, I'm just asking what would happen if we update this line

````
<Product type="0002" id="0002" name="TSP01 3 in 1 PIR Motion Sensor" config="wenzhou/tsp01.xml" />
````

to something like this line;

`````
<Product type="0002" id="0002" name="TSM2 4 in 1 PIR Motion Sensor" config="philio/pst02.xml" />
````

?

ualex73, 2016-01-06:
@Ertan:

We don't support duplicate id's. It was not smart of the vendor to re-use
the id's, because currently we can't differentiate between the 2 really
different devices.

At this moment we can't (and won't) change the TSP01 to TSM2, because it
'breaks' things for TSP01 users out there. As you mentioned, the TSM2 is a
subset of the TSP01, so the configuration should be roughly identical (so
you can configure what is supported). The values of the devices will work
fine, because those are discovered by OpenZWave and not configured.

If you really want it, you need to modify your local copy of openzwave to
have this - sorry about this.

ertanden, 2016-01-06:
Ok, I understand if it breaks TSP01. 

I was under the impression that the values reported are also in the config files, and since TSM02 is identical config wise to TSP01, but has an extra sensor value it wouldn't break TSP01.

Because as far as I remember when using the default OpenZwave configs I wasn't able to get separate Door/Window and PIR values (because TSP01 has only PIR). But later, when I changed the config to be of `philio/pst02.xml` then I could get separate Door/Window and PIR values. So, I was under the impression that there's something in config files that also affected reported values...

ualex73, 2016-01-06:
The values reported are 'discovered' by OpenZWave automatically, but it is
possible certain reporting requires (one time) configuration. Best to check
this in the manual.

Normally we use the configuration files for:
- configuration parameters
- association (group), e.g. to auto associate with controller (else no auto
reporting happens)
- remove invalid commandclasses, if the vender incorrectly implemented it
- tweak commandclasses, if the vendor incorrectly implemented it

Fishwaldo, 2016-01-09:
Like Alexie mentioned discovery of a devices capabilities is not done by config files but by the protocol. 

That being said, if they reuse ID's between devices:
1) the devices obviously did not go for certification with sigma/zwave alliance
2) hence potentially contain other bugs.

ertanden, 2016-01-11:
TSP01 is said to have certification number  ZC08-13050003 on their website.
http://www.tkbhome.com/?cn-p-d-258.html

Which is actually a Philio PSM02-1 in z-waveallinace.org
http://products.z-wavealliance.org/products/821/embedpics

And TSM02 has it's own certification number: ZC08-13050018
http://products.z-wavealliance.org/products/838/embedpics

PSM02-1 (aka TSP01) and TSM02 are identical as far as I know. Both 4-in-1 sensors. So I think TSP01 is just a repackage of the TSM02 without the magnetic contact part, hence they have the same ID. Even PSM02-1 has the same ID as TSP01 which confirms this.

So, I still believe we shall be good to replace TSP01 config file with content from `philio/psm02.xml`, which shall fully support TSM02 and also continue to be compatible with TSP01.

I took a diff of `philio/psm02.xml` and `wenzhou/tsp01.xml` and saw that the only difference is that TSP01 is missing the Door/Window contact mapping in the config file. If that's added, I think we shall have a config that works both for TSP01 and TSM02.

What do you guys think? I think you should be able to figure this out already but... anyways.

ben423423n32j14e, 2019-12-11:
Does anyone have a working config for this 4 in 1 sensor in the latest version of OpenZWave?

I used to be able to use the modifications specified here:

https://github.com/OpenZWave/open-zwave/pull/733/files#diff-b29cc5fd2e8a9b788117cc15a19b7094

FaberO42, 2022-05-13:
I too would like to know how to apply mods as noted here:
https://github.com/OpenZWave/open-zwave/pull/733/files#diff-b29cc5fd2e8a9b788117cc15a19b7094
Using ZWave to MQTT on Home Assistant.