OpenZWave/open-zwave #725: Can't include/exclude etc

sanderlv, 2016-01-05:
Trying to have OpenZwave running from withing Domotics. Domoticz points to OpenZwave:
http://www.domoticz.com/forum/viewtopic.php?f=47&t=7087&start=100

I can see the OpenZwave control panel succesfully but cannot include any device/exlcude whatever. Also the file OZW_Log.txt only shows:
2016-01-05 19:39:03.178 Always, OpenZwave Version 1.3.614 Starting Up

[image]

Fishwaldo, 2016-01-18:
You need to enable debugging. Please post your debug logs etc to the mailing list and not here.

closed 2016-01-18 by Fishwaldo

sanderlv, 2016-01-21:
Fishwaldo, not sure where the "Mailing" list is. thus I thought I'd copy paste it here:

2016-01-21 16:46:35.481 Always, OpenZwave Version 1.4.3 Starting Up
2016-01-21 16:46:35.481 Info, Setting Up Provided Network Key for Secure Communications
2016-01-21 16:46:35.481 Info, mgr,     Added driver for controller /dev/ttyACM0
2016-01-21 16:46:35.481 Info,   Opening controller /dev/ttyACM0
2016-01-21 16:46:35.481 Info, Trying to open serial port /dev/ttyACM0 (attempt 1)
2016-01-21 16:46:35.484 Info, Serial port /dev/ttyACM0 opened (attempt 1)
2016-01-21 16:46:35.484 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2016-01-21 16:46:35.485 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2016-01-21 16:46:35.485 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2016-01-21 16:46:35.485 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2016-01-21 16:46:35.485 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2016-01-21 16:46:35.485 Detail, 
2016-01-21 16:46:35.485 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2016-01-21 16:46:35.487 Detail, contrlr,   Received: 0x01, 0x10, 0x01, 0x15, 0x5a, 0x2d, 0x57, 0x61, 0x76, 0x65, 0x20, 0x33, 0x2e, 0x39, 0x39, 0x00, 0x01, 0x95
2016-01-21 16:46:35.487 Detail, 
2016-01-21 16:46:35.487 Info, contrlr, Received reply to FUNC_ID_ZW_GET_VERSION:
2016-01-21 16:46:35.487 Info, contrlr,     Static Controller library, version Z-Wave 3.99
2016-01-21 16:46:35.487 Detail, Node045,   Expected reply was received
2016-01-21 16:46:35.487 Detail, Node045,   Message transaction complete
2016-01-21 16:46:35.487 Detail, 
2016-01-21 16:46:35.487 Detail, contrlr, Removing current message
2016-01-21 16:46:35.487 Detail, 
2016-01-21 16:46:35.487 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x20) - FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2016-01-21 16:46:35.489 Detail, contrlr,   Received: 0x01, 0x08, 0x01, 0x20, 0xd3, 0xec, 0x71, 0xa5, 0x01, 0x3c
2016-01-21 16:46:35.489 Detail, 
2016-01-21 16:46:35.489 Info, contrlr, Received reply to FUNC_ID_ZW_MEMORY_GET_ID. Home ID = 0xd3ec71a5.  Our node ID = 1
2016-01-21 16:46:35.489 Detail, Node236,   Expected reply was received
2016-01-21 16:46:35.489 Detail, Node236,   Message transaction complete
2016-01-21 16:46:35.489 Detail, 
2016-01-21 16:46:35.489 Detail, contrlr, Removing current message
2016-01-21 16:46:35.489 Detail, 
2016-01-21 16:46:35.489 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x05) - FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2016-01-21 16:46:35.490 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x05, 0x28, 0xd7
2016-01-21 16:46:35.491 Detail, 
2016-01-21 16:46:35.491 Info, contrlr, Received reply to FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES:
2016-01-21 16:46:35.491 Info, contrlr,     There is no SUC ID Server (SIS) in this network.
2016-01-21 16:46:35.491 Info, contrlr,     The PC controller is a primary controller.
2016-01-21 16:46:35.491 Detail, Node215,   Expected reply was received
2016-01-21 16:46:35.491 Detail, Node215,   Message transaction complete
2016-01-21 16:46:35.491 Detail, 
2016-01-21 16:46:35.491 Detail, contrlr, Removing current message
2016-01-21 16:46:35.491 Detail, 
2016-01-21 16:46:35.491 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x07) - FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2016-01-21 16:46:35.494 Detail, contrlr,   Received: 0x01, 0x2b, 0x01, 0x07, 0x05, 0x02, 0x01, 0x15, 0x04, 0x00, 0x00, 0x01, 0xfe, 0x83, 0xff, 0x88, 0xcf, 0x1f, 0x00, 0x00, 0xfb, 0x9f, 0x7d, 0xa0, 0x67, 0x00, 0x80, 0x80, 0x00, 0x80, 0x86, 0x00, 0x00, 0x00, 0xe8, 0x73, 0x00, 0x00, 0x0e, 0x00, 0x00, 0x40, 0x1a, 0x00, 0x09
2016-01-21 16:46:35.494 Detail, 
2016-01-21 16:46:35.494 Info, contrlr,  Received reply to FUNC_ID_SERIAL_API_GET_CAPABILITIES
2016-01-21 16:46:35.494 Info, contrlr,     Serial API Version:   5.2
2016-01-21 16:46:35.495 Info, contrlr,     Manufacturer ID:      0x0115
2016-01-21 16:46:35.495 Info, contrlr,     Product Type:         0x0400
2016-01-21 16:46:35.495 Info, contrlr,     Product ID:           0x0001
2016-01-21 16:46:35.495 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_RANDOM: 0x01, 0x04, 0x00, 0x1c, 0x20, 0xc7
2016-01-21 16:46:35.495 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_INIT_DATA: 0x01, 0x03, 0x00, 0x02, 0xfe
2016-01-21 16:46:35.495 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_SET_TIMEOUTS: 0x01, 0x05, 0x00, 0x06, 0x64, 0x0f, 0x97
2016-01-21 16:46:35.495 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_APPL_NODE_INFORMATION: 0x01, 0x07, 0x00, 0x03, 0x01, 0x02, 0x01, 0x00, 0xf9
2016-01-21 16:46:35.495 Detail, Node002,   Expected reply was received
2016-01-21 16:46:35.495 Detail, Node002,   Message transaction complete
2016-01-21 16:46:35.495 Detail, 
2016-01-21 16:46:35.495 Detail, contrlr, Removing current message
2016-01-21 16:46:35.495 Detail, 
2016-01-21 16:46:35.495 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x56) - FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2016-01-21 16:46:35.496 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x56, 0x00, 0xac
[... cut ...]
2016-01-21 16:46:43.529 Detail, contrlr,   Received: 0x01, 0x07, 0x00, 0x4a, 0x1c, 0x06, 0x01, 0x00, 0xa9
2016-01-21 16:46:43.529 Detail, 
2016-01-21 16:46:43.529 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2016-01-21 16:46:43.529 Detail, Node006,   Expected reply was received
2016-01-21 16:46:43.529 Detail, Node006,   Message transaction complete
2016-01-21 16:46:43.529 Detail, 
2016-01-21 16:46:43.529 Detail, contrlr, Removing current message
2016-01-21 16:46:43.529 Detail, Notification: ControllerCommand - Completed

Fishwaldo, 2016-02-01:
https://groups.google.com/forum/#!forum/openzwave

2016-01-21 16:46:43.378 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2016-01-21 16:46:43.378 Info, contrlr, ADD_NODE_STATUS_FAILED

The controller reported a failure. It could be because its busy, interference etc etc etc. I've seen it a few times, if you keep trying, or move the devices around etc, it usually comes good.