OpenZWave/open-zwave #734: Config File Revision Numbers

Fishwaldo, 2016-01-11:
Implement a Revision Number for Config files and a mechanism to check for outdated config files, and update as needed (maybe via a 3rd party program?)

gizmocuz, 2016-02-11:
Thats great news! Would be great if this was build in the main library!
Or give a third party (the software application) an indication about an updated config, and the option to force the update on all or on separate nodes

Fishwaldo, 2016-02-15:
This will be build into the Library, with a API that will:
1) To Trigger Checks (or config option to check upon startup)
2) To provide a Notification Call Back about outdated Configs
3) with a option to Automatically download new versions
4) and refresh the device in OZW with the new config file.

in the case automatic lookups dont work:
1) A method to check config files for later revisions (for example, manual download) and perform (4) above. 

I'm considering a option to have 2 channels - Beta and Production as well.

Checks will be performed via DNS lookups, and the download will be via HTTP.

gizmocuz, 2016-02-15:
What do you mean by "Checks will be performed via DNS lookups", what if the machine does not have internet ?

bufferUnderrun, 2016-02-15:
Hi Justin,

Do you provide a way (ie: Option to ZWManager) to : 

- disable/enable this _AutoUpdate_ function  ?
- change the url where the config file are located to support custom (local) repositories ?

Thx

Fishwaldo, 2016-02-18:
> What do you mean by "Checks will be performed via DNS lookups", what if the machine does not have internet ?

The version number of the config file will be checked via a DNS lookup. If the machine doesn't have internet access - then proceed to:

> in the case automatic lookups dont work:
>1) A method to check config files for later revisions (for example, manual download) and perform (4) above.

Fishwaldo, 2016-02-18:
>    disable/enable this AutoUpdate function ?

yes - Via the options.xml file or Options Class. 

>    change the url where the config file are located to support custom (local) repositories ?

No - 2 reasons:
1) its backed by DNS, so there is some infrastructure required to push config files version numbers to DNS etc. I'm sure others can do that themselves. 
2) Most importantly - I'm trying to encourage upstreaming to OZW rather than hording it in individual applications. This way everyone benefits from updates rather than a individual application.

nechry, 2016-02-18:
I'm agree with you, (even I'm actually not good student with the delta we keep) if every application share update and new devices every one will benefits. A majority of my Zwave knowledge in very large part thanks to the openzwave library and it is the library that should benefit from my discoveries and improvements.

gizmocuz, 2016-02-18:
Okey, sure i must be missing something....

What i thought this was about is:
- You are using an (Aeotec) configuration file for a switch that has revision number 1 on a node
- after some time, OZW released an update for this configuration file with revision number 2
- the software or OZW detects a revision mismatch, and we can update the node's configuration file

As i am seeing it, no DNS lookup is needed.

Only the users zwcfg_xxxx.xml file needs to be checked against the configuration files

bufferUnderrun, 2016-02-18:
> No - 2 reasons:
> 2) Most importantly - I'm trying to encourage upstreaming to OZW rather than hording it in individual applications. This way everyone benefits from updates rather than a individual application. 

I'm agree with you. 

But in some organization, applications :
- can't connect to internet (unauthorized)
- connect via a proxy

My question sounds like a workaround for theses particular cases.

> in the case automatic lookups dont work:
> 1) A method to check config files for later revisions (for example, manual download) and perform (4) above. 

At last, if manual download is handled that's ok for me.

gizmocuz, 2016-02-18:
But the "config" folder with all configurations is always included with an application?
(And updated)

So we know the 'new' revision numbers, and the current once from zwcfg_xxxx.xml

fredolphin, 2016-05-27:
Was this great functionality finally implemented ? I can't find anything in the last release note related to this item.

Fishwaldo, 2016-06-01:
Its still being worked on. I'm doing some stuff on the backend at the moment (eg, website etc so its automated)

gizmocuz, 2016-06-05:
@Fishwaldo Do we need a website (online updating system), or will this also work stand-alone.
I mean, the driver can see the increase of a version number right ? (And update the main configuration file)

Fishwaldo, 2016-06-08:
@gizmocuz: The config files will be hosted on openzwave CDN network, and downloaded from there (http://download.db.openzwave.com)

There will be a config option to enable/disable the automatic update though, and it will just issue a "Notification" about outdated config files that the application can use to notify the user and they can manually update if they want.

the actual checking of the config file versions is done via DNS lookups:
eg:
`dig mfs.db.openzwave.com TXT @ns1.dynam.ac` gives you the current revision of the manufacturer_specific.xml file 
You could then download it:
http://download.db.openzwave.com/mfs.xml.<revision> or 
http://download.db.openzwave.com/mfs.xml

When you start up, it checks manufacturer_specific.xml version and downloads the latest version (if needed) plus any new config files. When new nodes are added/discovered, we check config file versions (if they exist) for those nodes, and download if outdated

All of the automatic checks can be disabled and manually triggered as well.

gizmocuz, 2016-06-09:
Ok, but how about offline systems ?

I thought that adding a simple revision number in each config file would do the trick.
Then you compare the revision number in the zwcfg_xxxx.xml file, against the original file, and update if needed?
That sound much easier, and does not need a CDN.
Also people modify/test the original configuration files (or implement new onces), this will not work with an online update system

Fishwaldo, 2016-06-09:
your scenario will work fine as well. If the Revision Number of config file does not match what was stored in zwcache_*.xml (renamed in the dev branch) then it will reload the node with the new config file.

For local changes - delete the zwcache_*.xml file for now and reload. If they are modifying local files, then disable the online updates.... 

The whole idea is to push out updated configs much more quickly here, without having to download tarballs etc. Make it as seemless as possible for the end user. 

There is still a need to distribute the updated config files.... hence the CDN...

Fishwaldo, 2016-08-18:
I'd like some feedback from you App Developers if possible:
If we detect a new config revision for a sleeping device, should we reload the device straight away, or issue a Notification and let the application trigger a reload? 

For always on devices, I'm assuming a automatic reload is fine. 

background - four ways we "get" new config revisions:
1) on Startup, we do checks on config revisions for Devices present on the network. Optionally, if out of date, download a new config file. (if download is disabled, then just a notification goes out)
2) on Startup, seeing a newer config file that what was recorded in zwcache_*.xml (manual edit of the config file, and restart)
3) on a application triggered call to check config files via DNS (eg, so application can periodically check for updates and download newer config files)
4) on a application triggered call to scan the config files for newer revisions (so users can edit files, and reload a node without restarting OZW/application)

1 & 2 are working in the Dev branch now, I just got to put some Manager hooks in for 3 and 4.

gizmocuz, 2016-08-19:
I would like to vote for 'Automatic Reload' when the device gets awake.
And maybe also a flag for a device (notification) that the config is out of date.
This because when the device will only be awake at certain time(slots), the user could be notified about this, and wakeup the device.

nechry, 2016-08-19:
This feature will also covert multi config files according to the device firmware version? We encounter more and more firmware update with new parameters, parameters size changes and different associations groups for same product type and id. If we can have a way, to choose (manually me be) to apply a specific revision on a device, in the same way as no 4.

Fishwaldo, 2016-08-19:
@gizmocuz sounds like a good idea. So a config setting with 3 options:
1) never - Manually triggered by app
2) Immediate - regardless of sleeping or not, reload
3) Awake - only reload when device wakes up.

@nechry nope. thats a different feature request. #644

nechry, 2016-08-19:
@Fishwaldo yes you right. I think now ,with this new way to update configuration the https://github.com/OpenZWave/open-zwave/issues/644 will be more easy to realise.

Fishwaldo, 2018-12-03:
This is Done.

closed 2018-12-03 by Fishwaldo