OpenZWave/open-zwave #739: **TODO: handle request for 0xa8** Please report this message.

slawekgh, 2016-01-12:
White USB Sigma for EU 
The same message for Danfoss Z and Fibaro Wall Plug 

Effect is 
a) there is no Power and Energy for FWP 
and
b) there is no heating for danfoss 

I can check also for termometers, motion detectors etc but imho doesnt make any sense - smth wrong with white USB ? 

the same devices (danfoss+fibaro_wall_plug) with black AEON USB Stick - everything works fine

slawekgh, 2016-01-13:
I made additional tests (to simplify I just run MinOZW only, nothing more):

For openzwave-1.0.791 and white Sigma USB Dongle:
a) without any devices (only one001) - OK, no MinOZW bad messages  
b) with Danfoss - MinOZW message: "Info, **TODO: handle request for 0xa8** Please report this message."
c) with Fibaro Wall Plug - "Info, **TODO: handle..." 
d) with Everspring Thermometer - "Info, **TODO: handle..." 
e) with all of them - "Info, **TODO: handle..." 

Exactly the same situation for:
openzwave-1.2.919
openzwave-1.4.0
openzwave-1.4.1

My system is Ubuntu 12.0 , I can also made some test on RaspberryPi with Raspbian 

sl

slawekgh, 2016-01-17:
Any ideas?

Fishwaldo, 2016-01-18:
No Idea. I don't know what message 0xa8 is at all. Could you send a complete log file to the mailing list?

I have a Sigma UZB stick (Gen 5) and it works fine.

slawekgh, 2016-01-19:
I ordered new USB Sigma stick , it is on the way , so please dont close this ticket, i will repeat tests asap and provide additional details , i think my USB stick could be damaged etc

slawekgh, 2016-02-16:
I have just received new (2nd one) white USB from sigma, I start to make tests...

slawekgh, 2016-02-16:
the same: "2016-02-16 21:55:56.144 Info, **TODO: handle request for 0xa8** Please report this message."

Fishwaldo, 2016-02-18:
Could you provide a full log please? 

I have updated UZB sticks and I don't see that message (ever!)

slawekgh, 2016-03-03:
[MINOZW.txt](https://github.com/OpenZWave/open-zwave/files/157678/MINOZW.txt)

slawekgh, 2016-03-03:
previous log file was for OZW 1.0.791 , now I send for OZW=1.4.1 
[MINOZW_1.4.1.txt](https://github.com/OpenZWave/open-zwave/files/157685/MINOZW_1.4.1.txt)

openzwave-dev-watcher, 2016-03-14:
How did you update your UZB?

Which tool do you use?

oteros2000, 2016-03-15:
I have the same problem like you,
 could you tell me if you have find the solution to work with lastest UZB?

Thanks.

slawekgh, 2016-03-15:
No, I didnt find solution, now I'm waiting for 5 new usb sticks and I plan to check it again on new sticks

IOOOTAlan, 2016-08-26:
I also have the same problem here with a SD5303 SoC and openzwave 1.4.78.
I reported the issue also on the [OpenZWave Google Group](https://groups.google.com/forum/#!topic/openzwave/iV1klDgeZQ0).
Guess this 0xA8 is a new kind of class in recent zwave controllers?

Fishwaldo, 2016-08-29:
Hi,
This is a message from the SerialAPI portion of the protocol, which is lower than the CC. Unfortunately we don't know what it means yet. 

I'm thinking it's either related to two things: initialization sequence isn't working for these newer chips, or maybe related to the new S2 Security support. I hope to investigate this further soon.

IOOOTAlan, 2016-08-29:
@Fishwaldo thanks for the info!

Fishwaldo, 2016-09-05:
Any way any of you can check what Firmware is loaded on the Z-Wave Chip?

Fishwaldo, 2016-09-06:
Hi,
This message is from Bridge Firmware loaded on your Chip/Controller. Currently OZW doesn't support Bridge Firmware. 

We only work with Static Controller Firmware. I'll see about adding a check in OZW to bail out if its not a Static Controller.

closed 2016-09-06 by Fishwaldo

IOOOTAlan, 2016-09-06:
Good to know, thanks for the analysis!

jeffsf, 2017-04-02:
Unfortunate that this has been closed as it is still a problem with a just-received UZB3-U-BRG from Sigma Design's authorized eval-kit distributor. 
* openzwave_1.4.164_armhf.deb
* libopenzwave1.3_1.4.164_armhf.deb
* libopenzwave1.3-dev_1.4.164_armhf.deb

Logs available, if needed.

Posts strongly that the BRG is a superset of the STA

From http://forum.z-wavepublic.com/t/uzb-static-versus-bridge/1041

> Hi, can someone explain the difference between ACC-UZB3--STA and ACC-UZB3--BRG. From the description, I see that the difference is static and bridge. Does that mean that bridge model will act as a repeater for all traffic and the static model will not? Thanks.
Answered with:

> The bridge controller allows the creation of virtual nodes. It is identical with the static controller in any other aspect.

Confirmed in http://forum.z-wavepublic.com/t/bridge-controller-uzb/22/2

>Regarding your questions:
> 1) Both are supported and have the same performance, while there is a slight difference in components.
> 2) While it is technically possible to update the static controller UZBs, however the firmware images and tools are not publicly available.
> 3) Z/IP Gateway, the core software component relies on the bridge functionality for session handling and IP associations.

> The onboard SAW filter has been replaced to match our latest recommendation for hardware designers.The Z-wave SoC is unchanged; the SD3503.

Fishwaldo, 2017-04-02:
We don't support Bridge Firmware. As the SerialAPI is not released by Sigma, not much we can do there unless someone wants to reverse engineer it.