OpenZWave/open-zwave #741: Additional TZ68 entry needed

philosifer, 2016-01-14:
An additional entry is needed for the TKB TZ68 as it looks like there are two versions with different IDs as can be seen in the database on pepper1.net.

New line is required in the manufacturer specific xml file as follows for type 3 and id 2.

 <Manufacturer id="0118" name="Wenzhou TKB Control System">
                <Product type="0003" id="0002" name="TZ68 On/Off Switch Socket" config="wenzhou/tz68.xml"/>

No other changes are needed as there is already a line in place defining the tz68 for type 101, id 103.

nechry, 2016-01-29:
@philosifer it's only this changes?
<Product type="00031" id="0002" name="TZ68E On/Off Switch Socket"/>

philosifer, 2016-01-29:
Yes. As far as I can see that's all. Maybe they just changed the type and id when it became Z-Wave+ compliant as that seems the only real difference

closed 2016-02-01 by Fishwaldo