OpenZWave/open-zwave #747: ZWNotification.Type.ValueChanged .net wrapper

marioeirea, 2016-01-18:
When a device is being polled, even if it's value has not changed, the notification with type ValueChanged is fired:

[image]

ValueRefreshed does not seem to ever fire if the when the node is polled.

Fishwaldo, 2016-01-18:
make sure "SuppressValueRefresh" is not set in any options.xml or application setting it via the Options class. 
OnValueRefreshed will only ever be triggered when SetChangeVerified is enabled for a ValueID. 

Any other questions, please post to the mailing list.

closed 2016-01-18 by Fishwaldo

marioeirea, 2016-01-19:
Correct me if I am wrong, but is the ValueChanged event supposed to fire only when the value of the node has changed not refreshed? Currently, every time the device is polled it fires an event that the value was changed even if it did not.

I made sure SuppressValueRefresh was not set.

Fishwaldo, 2016-01-19:
Please see the ValueChanged discussion on the mailing list yesterday