OpenZWave/open-zwave #748: hid.c regression on FreeBSD

stromnet, 2016-01-20:
With commit d78c7d08c95779c6ecaa4b846fbb600896f9fa9e the FreeBSD 10.2 support was broken again, as fixed in #701.

durin14/Augie Fackler: What FreeBSD version was that commit tested on? Commiter only stated clang version

stromnet, 2016-01-20:
Also, as we now have a proper release (v1.4), I'll go through with submitting my (not yet updated) freebsd-port upstream: https://github.com/stromnet/freebsd-ports/tree/master/devel/openzwave-devel

durin42, 2016-01-20:
I'm on 10.1-RELEASE right now.

stromnet, 2016-01-20:
Okay. I *think* the preferred solution is to install converts/iconv and link against that:

https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=199099#c37

At least thats how I interpret it... Will have to look into a bit more.

durin42, 2016-01-20:
Whee. We should probably document the installation process for FreeBSD if that's the case so others don't get confused. :) Thanks for digging!

stromnet, 2016-01-20:
Once the port is accepted upstream, it would be `pkg install openzwave-devel`.. :)

durin42, 2016-01-20:
I don't know that that'll work with the way python-openzwave wants to be compiled. :(

stromnet, 2016-01-20:
Should probably be doable, unless python-openzwave does something very weird.
If you can you point to some instructions, I'll take a quick look at that as well when I get around to this.

durin42, 2016-01-21:
https://github.com/OpenZWave/python-openzwave/blob/master/INSTALL_ARCH.txt is the instructions for python-openzwave. I've done it a couple of times in a jail, but I never tried to install openzwave ahead of time and then had python-openzwave use that set of libraries, I just let it statically link it all together.

(For my particular use case, I also need it for Python 3, and I can't remember if there's anything special that needs to happen for ports for that. By all means drop me an email or private message if you want testing or anything.)

Fishwaldo, 2016-01-21:
So is this still a issue? Does that commit need to be reverted or ???

xbeaudouin, 2016-01-21:
@stromnet You are makeing a openzwave port ? Tell me when it is finished, I would like to link domoticz on it (was makeing one...)

stromnet, 2016-01-21:
@Fishwaldo: Yes, it became an issue again.. :) FreeBSD 10.1 base iconv still uses static modifier. FreeBSD 10.2 base iconv, and iconv from ports, uses non-static. So simple solution is probably just to build with system iconv. Will look into the best solution.

@xbeaudouin: I have a working -devel port, i.e. github head from whenever. I haven't updated this regularly, just now and then. Now that we have 1.4, I'll create a proper port and submit it upstream. I'll report progress here!

@durin42: *should* not be a problem, but we'll see.. :)

xbeaudouin, 2016-01-21:
@stromnet Would like to help if you want :)

stromnet, 2016-01-21:
@xbeaudouin or @durin42, can you please help me test https://github.com/stromnet/open-zwave/tree/iconv-fix on 10.1 (and/or earlier)?

It reverts the hid.c patch to be posix-normative, and the (g)makefile should now ask pre 10.2 users to install libiconv, if not installed. Hopefully works on all FreeBSDs!

I've updated the openzwave-devel port now, almost also the openzwave main port, but note sure if it should sit in devel/ or comms/...

xbeaudouin, 2016-01-22:
@stromnet Works for me. Will adapt my domoticz port to hande this new port. Ping me when it is commited :)

stromnet, 2016-01-22:
@xbeaudouin: great! Which FreeBSD version did you test this on, and did you test the port or just the open-zwave build?

xbeaudouin, 2016-01-22:
I am using on 10.2, but I don't yet a Zwave stick to test (have to buy one btw) - moving all stuff at home into domoticz -

stromnet, 2016-01-22:
Okay, let's see if @durin42 can confirm that it builds fine on 10.1 too then. It *did* build fine in my poudriere jail but.. better to test on a real system.

xbeaudouin, 2016-01-22:
My domoticz port is BTW on my github : https://github.com/xbeaudouin/freebsd-ports

durin42, 2016-01-22:
Will test this tonight on my 10.1 box.

stromnet, 2016-01-23:
Any luck? :)

durin42, 2016-01-23:
Your branch compiled just fine on 10.1 just now with no funny business. Thanks!

stromnet, 2016-01-23:
Great! Opening a PR then!

durin42, 2016-01-23:
Thanks for all the work on this! I look forward to snagging these libs from ports some day. :)

stromnet, 2016-01-23:
Submitted!

https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=206526

Now it's up to the FreeBSD port management to accept and include it. From experience, this can take anything from a few days to months...

PR #750 (2016-02-01): Multi-platform compatible FreeBSD build fix for iconv
Revert hid.c change in d78c7d0, and enforce older FreeBSD versions to
use libiconv from ports instead of old non-posix from base.

Fixes #748, tested on both 10.1 and 10.2. Should work on 9.3 too.
commit af7b40c7 (2016-02-01): Multi-platform compatible FreeBSD build fix for iconv

Revert hid.c change in d78c7d0, and enforce older FreeBSD versions to
use libiconv from ports instead of old non-posix from base.
Should fix #748
closed 2016-02-01 by Fishwaldo