OpenZWave/open-zwave #749: Old EZMotion HSM100 not working if setasreport="true"

ugoviti, 2016-01-21:
OZW Version: 1.4-3-g3f1d609

I discovered that my old HSM100 Wireless Multi-Sensor (2011/2012 version)  from EZMotion doesn't works if in the file homeseer/hsm100.xml is enabled:

<CommandClass id="32" setasreport="true"/>

I have two EZMotion HSM100, are recognized as "Homeseer HSM100 Wireless Multi-Sensor":

        <Manufacturer id="1e" name="Homeseer">
            <Product type="2" id="1" name="HSM100 Wireless Multi-Sensor" />
        </Manufacturer>

OZW_Log when setasreport="true" is used (it doesn't work):
=====================================
2016-01-21 08:28:20.466 Detail, Node058,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x3a, 0x03, 0x20, 0x01, 0xff, 0x15
2016-01-21 08:28:20.467 Detail, 
2016-01-21 08:28:20.467 Info, Node058, Received Basic set from node 58: level=255. Treating it as a Basic report.
2016-01-21 08:28:20.480 Detail, Node058,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x3a, 0x03, 0x20, 0x01, 0xff, 0x15
2016-01-21 08:28:20.481 Detail, 
2016-01-21 08:28:20.481 Info, Node058, Received Basic set from node 58: level=255. Treating it as a Basic report.
2016-01-21 08:28:20.506 Detail, Node058,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x3a, 0x03, 0x20, 0x01, 0xff, 0x15
2016-01-21 08:28:20.507 Detail, 
2016-01-21 08:28:20.507 Info, Node058, Received Basic set from node 58: level=255. Treating it as a Basic report.
2016-01-21 08:28:20.535 Detail, Node058,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x3a, 0x03, 0x20, 0x01, 0xff, 0x15
2016-01-21 08:28:20.536 Detail, 
2016-01-21 08:28:20.536 Info, Node058, Received Basic set from node 58: level=255. Treating it as a Basic report.
2016-01-21 08:28:20.547 Detail, Node058,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x3a, 0x03, 0x20, 0x01, 0xff, 0x15
2016-01-21 08:28:20.547 Detail, 
2016-01-21 08:28:20.547 Info, Node058, Received Basic set from node 58: level=255. Treating it as a Basic report.
2016-01-21 08:28:20.567 Detail, Node058,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x3a, 0x03, 0x20, 0x01, 0xff, 0x15
2016-01-21 08:28:20.567 Detail, 
2016-01-21 08:28:20.568 Info, Node058, Received Basic set from node 58: level=255. Treating it as a Basic report.
2016-01-21 08:28:20.643 Detail, Node058,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x3a, 0x03, 0x20, 0x01, 0xff, 0x15
2016-01-21 08:28:20.644 Detail, 
2016-01-21 08:28:20.644 Info, Node058, Received Basic set from node 58: level=255. Treating it as a Basic report.
=====================================

OZW_Log when setasreport="true" is not used (it works):
=====================================
2016-01-21 21:49:59.743 Detail, Node058,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x3a, 0x03, 0x20, 0x01, 0xff, 0x15
2016-01-21 21:49:59.744 Detail, 
2016-01-21 21:49:59.745 Info, Node058, Received Basic set from node 58: level=255.  Sending event notification.
2016-01-21 21:49:59.746 Detail, Node058, Notification: NodeEvent
=====================================

What can I do to solve that problem?

Disabling setasreport="true" upstream can make other version of that sensor do not works?

Thank you for the help.

With kind regards

nechry, 2016-01-22:
can you try this version:
https://github.com/jeedom/plugin-openzwave/blob/master/ressources/openzwave/config/homeseer/hsm100.xml

I recently work on this device and now I can use it. but It's need a little trick, on NodeEvent notification I interpreter as a Basic CC with this I receive detection event.

my devices are in application version: 2.01, 2.02

ugoviti, 2016-01-26:
Hi Nechry,

I confirm that your hsm100.xml version is working correctly now.

so we must use:

    <CommandClass id="32" setasreport="add"/>

instead:

    <CommandClass id="32" setasreport="true"/>

Can you make pull request to make upstream?

With kind regards

Fishwaldo, 2016-01-27:
hi,
setasreport="add" isn't valid. Just remove it instead.

nechry, 2016-01-29:
my config file contains
<CommandClass id="32" action="add" /> not setasreport="add"
this config file will merged very soon.

Fishwaldo, 2016-02-01:
Dropping this as setasreport="add" is invalid.

closed 2016-02-01 by Fishwaldo

nechry, 2016-02-01:
the jeedom config file don't contains setasreport="add" but

<CommandClass id="32" action="add" />
see:
https://github.com/jeedom/plugin-openzwave/blob/master/ressources/openzwave/config/homeseer/hsm100.xml#L97

ugoviti, 2016-02-01:
my mistake too, I meant:

    <CommandClass id="32" action="add" />