OpenZWave/open-zwave #754: Philio PAN08-1 In Wall Roller Shutter Controller (new configuration file)

pbda, 2016-01-26:
Hello,

Attached config file for Philio PAN08-1 In Wall Roller Shutter Controller. Exists also as Zipato PAN08-1.

    `# diff manufacturer_specific.xml.ref manufacturer_specific.xml
    549c549
    < 		<Product type="0001" id="0006" name="PAN08-1 In Wall Roller Shutter Controller"/>
    ---
    > 		<Product type="0001" id="0006" name="PAN08-1 In Wall Roller Shutter Controller with Power Measurement" config="philio/pan08.xml"/>`

[pan08.xml.txt](https://github.com/OpenZWave/open-zwave/files/105564/pan08.xml.txt)

Fishwaldo, 2016-02-01:
There is no Association Information in the config file. Please add it:
https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

pbda, 2016-02-01:
Documentation says:
> There is only one group called Group1, there is only one node for
> Group1 which support MULTILEVEL_SWITCH_REPORT, METER_
> REPORT_COMMAND_V3, ALARM_REPORT

I am not sure but:
```
<CommandClass id="133">
  <Associations num_groups="1">
    <Group index="1" max_associations="1" />
  </Associations>
</CommandClass>
```
should be OK.
I join modified file: [pan08.xml.txt](https://github.com/OpenZWave/open-zwave/files/112365/pan08.xml.txt)

Fishwaldo, 2016-02-05:
Almost correct. You forgot the label. I added it for you. :)

closed 2016-02-05 by Fishwaldo