OpenZWave/open-zwave #757: Working config file for Qubino ZMHNID1 ZWave+ thermostat

stephaneguillard, 2016-01-30:
Hello,

Please add this file to Config/qubino

[ZMNHID1.zip](https://github.com/OpenZWave/open-zwave/files/110862/ZMNHID1.zip)

Regards,
-- 
Stéphane

Fishwaldo, 2016-02-01:
please check the updated Qubino Config files just commited and let us know if there needs to be further changes.

nechry, 2016-02-01:
The news qubino or the olds config files have, from my point of view, the same behavior in openzwave. All issues still presents and are related to multiInstance / MultiChannel Qubino implementation.

stephaneguillard, 2016-02-01:
Hello,

I've seen these new files (let's call them 'A') while I was working on mine ('B').

The text labels in the 'A' files are:
- either directly taken from less-than-perfect Qubino documentation written in English by slovenians, whcih often renders them pointless (what is 'Too low temperature limit'? should be 'Low temperature alarm threshold')

- or written in french-glish (what is 'fil pilote'? what is 'interrupteur' etc.) and often misleading

Beyond that, the 'A' files won't work. Examples for the ZMNHID1, for which I created and successfully tested the file 'B' file:

- the 'A' file removes COMMAND_CLASS_BASIC. This renders the thermostat useless as it forbids the application to switch its operating mode between Off and Auto, and the device is preset with an Off factory setting.

- the 'A' file does not define groups 4 and 5, which are used by the device to broadcast low and high temperature alarms, as set per parameters 60 and 61.

As of now, I have only proposed two 'B' files:
- ZMNHJD1.xml on January 8
- ZMNHID1.xml on January 31

So I kindly ask you to replace the corresponding 'A' files by mine, as their text labels are written in valid English, and their features are tested.

Side note: the reason why many ZMNHIA2 users complain about the fact that it won't work with OpenZWave-based home automation software is that the ZMNHIA2.xml also removes COMMAND_CLASS_BASIC, rendering the thermostat useless as it forbids the application to switch its operating mode between Off and Auto, and the device is preset with an Off factory setting.

I am in touch with Qubino, and willing to help refine and fix all these files, assuming my changes and new files are not overwritten by buggy, non-tested ones with broken English labels.

Regards,
-- 
Stéphane

nechry, 2016-02-01:
Yes I found also many wrong definition I updated 2 weeks ago the mixed language content in jeedom repo.

For COMMAND_CLASS_BASIC removed, I think all config contains this entry and maybe is a root cause of some other problems we have.

stephaneguillard, 2016-02-04:
So Justin, you asked me to let you know if there needs to be further changes, I told you so:)

Fishwaldo, 2016-02-05:
Sorry if you feel offended, but taking a config file from a manufacturer is 'hopefully' a bit more reliable. Neverless, now that I hear your changes, I'll accept them. 

> Side note: the reason why many ZMNHIA2 users complain that it doesn't work with OpenZWave-based home automation software is that the ZMNHIA2.xml also removes COMMAND_CLASS_BASIC, rendering the thermostat useless as it forbids the application to switch its operating mode between Off and Auto, and the device is preset with an Off factory setting.

If this really is the case, then the device is a lot more broken than I thought. :) 
Basic CC is meant to be mapped to another CommandClass, and if you read the writeups/groups on the Basic CC, you will see its more intended as a "interopability" command class (between devices of different types) than a "controllable" command class.

Fishwaldo, 2016-02-05:
`Justin:open-zwave justinhammond$ make xmltest
...
/Users/justinhammond/Development/open-zwave/config/qubino/ZMNHID1.xml:199: parser error : Comment must not contain '--' (double-hyphen)
	
	    ^
make: *** [xmltest] Error 1
`
Please ensure you run "make xmltest" when updating/submitting config file updates.

closed 2016-02-05 by Fishwaldo

stephaneguillard, 2016-02-05:
There's no offense at all:) I was just trying to fix broken bits.

jotakar, 2016-05-16:
I've a problema with this device. I deleted association with node 1 (lifeline). I can set any other group with any node, except lifeline. I can't associate group 1 of this device to anyone.
Also it meters power with an error (it's controlling a resistor of 600 watts and read just about 100 w).
It sound's very strange. It ws working fine until apair of days.
And of course it Works: turn ON/OFF, reads temperatura, report power (with manual refresh). 
I have no idea about the solution.

stephaneguillard, 2016-05-23:
Hello,

Won't comment on the conséquences of deleting the lifeline association.

As to the erroneous power report: I have 6 of these ZMNHID1 modules at home, they all work flawlessly but one of them had to be replaced precisely because it reported ~100 watt when driving a 600W resistive load.

So I suggest you RMA it to your vendor.

Regards,
-- 
Stéphane