OpenZWave/open-zwave #765: Manufacturer, product and type Ids default value is troublesome

dsoulayrol, 2016-02-03:
Hi everyone,

In `Node.cpp`, default values for `m_manufacturerId`, `m_productType` and `m_productId` is `0`. However, `0` seems to be a valid value for each one of those attributes, even if it's true that no device in `config/manufacturer_specific.xml` is currently identified by this very 3-uplet.

I suggest that those members should be initialized with some value that seem to never be used, like `0xffff`.

Additionally, documentation is false when it states that *an empty string* is returned if the *specific command class is not supported by the device*. Actually, `Manager::GetNodeManufacturerId` can return `Unknown` and `Driver::GetNodeManufacturerId` can return `0` by itself.

Fishwaldo, 2016-02-05:
Changing that would probably break a lot of applications, so I wont do it.... and as you mentioned, no one has been assigned a "0" manufacturer ID, and I doubt anybody will do a type or product id of 0.

> Additionally, documentation is false when it states that an empty string is returned if the specific command class is not supported by the device. Actually, Manager::GetNodeManufacturerId can return Unknown and Driver::GetNodeManufacturerId can return 0 by itself.

I'm lost, the two functions you reference are not command class related, but manufacturer related?

closed 2016-02-05 by Fishwaldo

dsoulayrol, 2016-02-05:
> I'm lost, the two functions you reference are not command class related, but manufacturer related?

Sorry. I meant that the following were inexact or incomplete. I found cases in the code that would return `Unknown` for example.

- http://www.openzwave.com/dev/classOpenZWave_1_1Manager.html#a43cb00a5a6fabad3ffbb178269fe0b27
- http://www.openzwave.com/dev/classOpenZWave_1_1Manager.html#a9d5575157b03ef818a13d3b574362c60
- http://www.openzwave.com/dev/classOpenZWave_1_1Manager.html#a524fc5ac8bbd2226ac35e289ffc3c2f6

> Changing that would probably break a lot of applications, so I wont do it.

I understand and I now handle this case accordingly. However, I think this case should deserve a warning in the aforementioned documentation entries.

Thanks.