OpenZWave/open-zwave #771: Header files in subfolders have incorrect includes

julienw, 2016-02-08:
Most headers include other headers using `#includes "Header.h"`. But the quote includes look in the same directory, and in directories specified by -I then.

But the headers in the subdirectories (eg: value_classes, or platform), that most include other open-zwave header using eg `#include "Defs.h"` fail when we don't use `-I path-to-openzwave-headers` to build.

This is too bad because this could be easy to fix by using relative paths instead.

julienw, 2016-02-08:
Includes like `#include "value_classes/Value.h"` in `value_classes/ValueList.h` yield the same issue.

julienw, 2016-02-08:
Note this is especially useful with the headers installed globally (for example using the debian `-dev` package). If this bug is fixed, we can just use `#include <openzwave/Options.h>" etc and we don't need any `-I` options, which feels right to me.

Fishwaldo, 2016-02-10:
I'm well aware of the benifits, but this  would break existing builds for a lot of 3rd party apps that use OZW and give instructions to their users how to compile etc. 

Read the mailing lists/other issues for previous discussions on it. The change wont happen in a 1.x timeframe

closed 2016-02-10 by Fishwaldo

julienw, 2016-02-10:
Actually there are 2 different issues here. Let me try to explain.

1. The "main" header files (eg: Options.h, Manager.h). For these ones you need -I if they're not in the usual locations. And... why not. It would be a lot easier if the repo's layout would have the installed layout (ie: a `openzwave` subdirectory where the headers are). But I agree this is your choice.

2. The subheader files that are not used directly by clients (the ones in the subfolders `value_classes`, or `platform`, etc). These ones have incorrect includes, and using `#include "../Defs.h"` (for example) would just work without needing a -I clause, because the preprocessor is looking relatively from the including file.

Fishwaldo, 2016-02-10:
1) It's been this way since the inception of OZW. Not my choice. 
2) as mentioned, changing any include files/paths will break compiles for dozens of third party Apps based on OZW and make me unpopular with those authors 
3) you can always use pkg-config (or ozw_config) to setup the right include paths for your Makefiles etc, which means even if users install in weird paths (like opt) then it still works.

julienw, 2016-02-11:
I'm focussing on the issue 2. here. What I'm saying is that changing these should have zero impact on users.

Currently, `#include "Defs.h"` inside (for example) `value_classes/ValueID.h` works only because we use `-I <path to openzwave includes>`. If we'd use `#include "../Defs.h"` this would work in all cases (because this would be resolved _first_ using the `value_classes` directory as starting point). No need to change anything in how current users are building, yet less surprise to new users.

vexofp, 2016-02-17:
Indeed. Changing #include's internal to the different headers should have no effect on how user code can include them. This change only _removes_ an implicit dependency on the users build environment being set up a particular way.

The current recommended setup of adding the openzwave root to the include path and including headers directly will continue to work, but the option of not doing so and including necessary headers by their full path will also be enabled.

Fishwaldo, 2016-02-18:
Honestly, I feel its a "change" just because it can be a change. Both ways work. No way is superior to the other way with the "simple" change, and there is no risk to breaking things on weird configs by not changing. 

Not to say the header files are perfect, but if it aint broken, there is nothing to fix!

As I mentioned, we can look at it when we start considering 2.x, but right now, I'd rather spend time on adding new features than working on something that has no additional benefit to the end users!

julienw, 2016-02-18:
Well, with the current way, I need to do this: https://github.com/fxbox/openzwave-rust/blob/master/openzwave-sys/build.rs#L10-L1, basically "guess" where the include files will be :/

I agree this is something we can workaround and thus is not the highest priority. But if somebody would contribute this, would you take it ?

hfiguiere, 2016-03-02:
You could actually fix the .pc file to reference the proper path....

hfiguiere, 2016-03-02:
See pull request above.

julienw, 2016-03-03:
IMO this is not fixing the root issue.

hfiguiere, 2016-03-03:
It is providing an automatic work around. Use pkg-config and you have the right includes setup. 

Looking at the previous situation it was miraculously working if you didn't have openzwave installed in a fancy prefix.

So it is still better than before.

julienw, 2016-03-03:
Actually with the current situation it's not miraculously working even if openzwave is installed in a proper place. That's why I want the headers to be fixed. Yet both changes may be needed to make it work correctly. Your change may be simpler to make it "just work" though.

One caveat: the .pc file is not installed with the debian packages, you may want to fix the debian scripts too.

julienw, 2016-03-15:
I created https://github.com/OpenZWave/open-zwave/pull/802 to include .pc in the Debian package.