OpenZWave/open-zwave #775: Crash Dot Net example on OZW 1.4.1

elcosmelco, 2016-02-14:
Hi,

I've just updated from OZW 1.3.496 to 1.4.1, and my first course of action is usually compile and run the DotNet example and generate a config file.
When I run this, after about 15 minutes (I have about 45 nodes) it gets to a Fibaro RGBW node and visual studio dies without exceptions or callstack (vshost.exe crashes).

I'm assuming COMMAND_CLASS_COLOR is the guilty party here because I've run the build 3 times and it is always the last thing mentioned in the log:

_CommandClass: COMMAND_CLASS_COLOR, RequestFlags: 0, Instance: 1, Index: 255
_
The log is shared here:
[(http://www.openzwave.com/OZWLog.log?wtd=2uwArnAF6iAbHTfe&request=resource&resource=o6y512t&rand=43)]

(I get a lot of phantom nodes there - my last real NodeId is 65)

This is my first post / issue on github - so if I've done something wrong please let me know.

Cheers,
Elco

Fishwaldo, 2016-02-15:
Have you tried building the entire project in Debug Mode (the OZW library etc)

We really need a backtrace to diagnose whats going on. I know a few projects using the dotNet Wrapper with the COLORCOMMAND_CLASS so I assume its just something up with the dotNet example App.

[(http://www.openzwave.com/OZWLog.log?wtd=2uwArnAF6iAbHTfe&request=resource&resource=o6y512t&rand=43)]

elcosmelco, 2016-02-15:
Yeah the solution contains all projects (OZW, the .Net lib and the example app) and they're all set to be built in Debug Mode but when it crashes there's just a popup saying svchost.exe has crashed, nothing in the debugging session of the app itself. Anything I could try?

Cheers,
Elco

Fishwaldo, 2016-02-18:
I know nuts about windows platforms (enough to get by compiling) so I can't really help with "how to"

elcosmelco, 2016-02-21:
The issue doesn't seem to be present in the latest snapshot (1.4.53) so I suppose you can close this issue.

Cheers,
Elco

closed 2016-02-24 by Fishwaldo