OpenZWave/open-zwave #781: AssociationCmd_Get can loop indefinitely on startup

dnpwwo, 2016-02-19:
I had 28 devices in my network including some Aeotec Micro Smart Energy Dimmer v2 devices that were functioning okay with OpenZWave but after adding another 18 devices (to a total of 46) my system wouldn't start anymore.  Further investigation showed that the first Dimmer (Node038) was no longer getting through the associations step of the start up (although it had for the last 6 months).

I've attached the log for the device here and my reading of it is:

- Device reports that it has 2 groups
- First group is queried and it has one node
- OpenZWave asks for the 2nd group and the device responds with details of the 1st group again
- OpenZWave asks for the 2nd group and the device responds with details of the 1st group again
- OpenZWave asks for the 2nd group and the device responds with details of the 1st group again
- you get the idea....

[Node038.zip](https://github.com/OpenZWave/open-zwave/files/138851/Node038.zip)

The current code works out the next group to query from the previous response which, while elegant, makes the process less robust than it could be.  I worked around this locally by added a 'NextGroup' member variable to the Associations class and incrementing that instead of trusting devices to respond correctly.

(Big fan of OZW BTW, long live Open Source!)

nechry, 2016-02-20:
I got from time to times the same issues, always with a NodOn ASP-3-1-00 Smart Plug

Fishwaldo, 2016-02-21:
Seems like a reasonable fix. Could you post a patch/pull request?

nechry, 2016-02-21:
be carfull with some devices skip index position like the Qubino ZMNHTAx Smart Meter
https://github.com/jeedom/plugin-openzwave/blob/master/ressources/openzwave/config/qubino/ZMNHTAx.xml
```
<!-- Association Groups -->
	<CommandClass id="133">
		<Associations num_groups="5">
			<Group index="1" max_associations="1" label="Lifeline" auto="true"/>
			<Group index="2" max_associations="16" label="Basic on/off (IR external relay)"/>
			<Group index="3" max_associations="16" label="Switch binary report (IR external relay)"/>
			<Group index="5" max_associations="16" label="Basic on/off (External relay)"/>
			<Group index="6" max_associations="16" label="Switch binary report (External relay)"/>
		</Associations>
	</CommandClass>
```
index number 4 is not present.

nechry, 2016-10-06:
I also recently get this issue with a Fibaro WallPlug

quarcko, 2019-07-16:
Also Neo Coolcam Siren Alarm does same thing now.
I don't know what changed in my setup as it was working perfectly for about 8 months from install
and only now started to exhibit this behavior.

What i spotted in my Z-Wave network map is that when everything was working normally
Neo Coolcam was not direct child of a Aoetec stick (it was behind another device)

And recently i physically moved Neo Coolcam to another location and on the map
it became directly connected to the stick. (maybe causing this issue?)

andzie82, 2020-11-23:
Hi, any update on this or is there a workaround because my zwave network is to busy with this issue according to the logs,
https://pastebin.com/LwttF0HU

RicardP, 2020-11-29:
@andzie82 Don't expect anything from this project, rarely any improvements. OZW is controlled by one person that seems not mind that thousands of people did put their trust and money into ZWave equipments because of OZW exists.

Fishwaldo, 2020-11-30:
@RicardP wow. This is the type of attitude that demotivates me from doing anything on this project. Banned...