OpenZWave/open-zwave #783: Device Configuration with many end points

nechry, 2016-02-23:
I just discover a device with 5 end points in configuration.
the Horstmann SES303
http://www.vesternet.com/downloads/dl/file/id/998/product/2098/z_wave_secure_ses302_ses303_temperature_humidity_sensors_user_and_installation_manual.pdf
the config file is:
https://github.com/jeedom/plugin-openzwave/blob/master/ressources/openzwave/config/horstmann/ses303.xml

In ozw lib I found this:
https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/command_classes/Configuration.cpp#L225

Is the first time I see more of one instance for configuration.

It is something lib ozw will support ?

Fishwaldo, 2016-02-24:
I've never seen that before. There should be no reason why it wouldn't work but we need to update the Configuration CC to support multiinstances. 

If I craft a patch can you test?

http://www.vesternet.com/downloads/dl/file/id/998/product/2098/z_wave_secure_ses302_ses303_temperature_humidity_sensors_user_and_installation_manual.pdf <http://www.vesternet.com/downloads/dl/file/id/998/product/2098/z_wave_secure_ses302_ses303_temperature_humidity_sensors_user_and_installation_manual.pdf> 
https://github.com/jeedom/plugin-openzwave/blob/master/ressources/openzwave/config/horstmann/ses303.xml

https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/command_classes/Configuration.cpp#L225

nechry, 2016-02-24:
I don't have the device, but I can check to got one or find another solution to test. But I will need also to update the python wrapper or a update of the controlpanel to test the patch.

Fishwaldo, 2016-02-25:
The wrappers should need to be updated as the config for the additional instances will just be exposed as additional ValueID's. 

Apps might need to be updated depending on how they handle config ValueID's but ozwcp should work without updates. 

The question is, in this case, if those additional sensors are not hooked up, does the device still expose the config values for the instances that map to the those sensors. If not, then we are going to have timeouts....

Fishwaldo, 2020-03-05:
According to the Specifications, Configuration CC can only be on the Root Endpoint.

closed 2020-03-05 by Fishwaldo