OpenZWave/open-zwave #787: Fix Command_Class_Alarm everspringct/hsm02.xml

Nico0084, 2016-03-03:
Hi
 There is some errors in HSM02 file config,
Command class alarm is on version 1 and device isn't a multi-instance
See attached file hsm02.xml.txt to fix it
[hsm02.xml.txt](https://github.com/OpenZWave/open-zwave/files/156431/hsm02.xml.txt)

I propose another solution to fix that with a ValueTypeList for "Alarm Level"
But we should modify ozw code, and I not sure of other effect in it.
Command_class_alarm create byte values, and in this case we should remove one and replace it with type list.
On my proposal, this is doing during void Node::ReadValueFromXML.
If there is diffence type between  Value created with command_class and Value from xml config, replace process is do.
This must ensure that xml file have no error, otherwise I'm not sure of side effect !

But with this replacement we could have a better integration and much flexibility to handle device, not ?

see git diff of alarm.cpp and node.cpp  and new HSM02 xml : 
```
diff --git a/cpp/src/command_classes/Alarm.cpp b/cpp/src/command_classes/Alarm.cpp
index 3dad19c..c2934eb 100644
--- a/cpp/src/command_classes/Alarm.cpp
+++ b/cpp/src/command_classes/Alarm.cpp
@@ -229,12 +229,30 @@ bool Alarm::HandleMsg
                        value->OnValueRefreshed( _data[1] );
                        value->Release();
                }
-               if( (value = static_cast<ValueByte*>( GetValue( _instance, AlarmIndex_Level ) )) )
+               // For device on version 1 the level could have different value. This level value corres
+               if ( Value* value = GetValue( _instance, AlarmIndex_Level ) )
                {
-                       value->OnValueRefreshed( _data[2] );
+                       switch ( value->GetID().GetType() )
+                       {
+                               case ValueID::ValueType_Byte:
+                               {
+                                       ValueByte* valueByte = static_cast<ValueByte*>( value );
+                                       valueByte->OnValueRefreshed( _data[2] );
+                                       break;
+                               }
+                               case ValueID::ValueType_List:
+                               {
+                                       ValueList* valueList = static_cast<ValueList*>( value );
+                                       valueList->OnValueRefreshed( _data[2] );
+                                       break;
+                               }
+                               default:
+                               {
+                                       Log::Write( LogLevel_Info, GetNodeId(), "Invalid type (%d) for A
+                               }
+                       }
                        value->Release();
                }
-
                // With Version=2, the data has more detailed information about the alarm
                if(( GetVersion() > 1 ) && ( _length >= 7  ))
                {

```

```
diff --git a/cpp/src/Node.cpp b/cpp/src/Node.cpp
index 926933d..475b0e1 100644
--- a/cpp/src/Node.cpp
+++ b/cpp/src/Node.cpp
@@ -2545,8 +2544,16 @@ void Node::ReadValueFromXML
        {
                if( Value* value = store->GetValue( id.GetValueStoreKey() ) )
                {
-                       value->ReadXML( m_homeId, m_nodeId, _commandClassId, _valueElement );
-                       value->Release();
+                       // Check if values type are the same
+                       ValueID::ValueType v_type = value->GetID().GetType();
+                       if ( v_type == type ) {
+                               value->ReadXML( m_homeId, m_nodeId, _commandClassId, _valueElement );
+                               value->Release();
+                       } else {
+                               Log::Write( LogLevel_Info, m_nodeId, "xml value type (%s) is different t
+                               store->RemoveValue( value->GetID().GetValueStoreKey() );
+                               CreateValueFromXML( _commandClassId, _valueElement );
+                       }
                }
                else
                {

```

```
			<CommandClass id="113" name="COMMAND_CLASS_ALARM" version="1" request_flags="2" innif="true">
				<Value type="byte" genre="user" instance="1" index="0" label="Alarm Type" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="12">
					<Help>Grouping 1, afterward, LED will light up for 1 sec to remind user when HSM02 is triggered due to open or close incidents.</Help>
				</Value>
				<Value type="list" genre="user" instance="1" index="1" label="Alarm Report" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" vindex="0" value="0" size="1">
					<Help>
						No alarm (0) : There is no alarm to report. 
						Power applied (2): Once the power has been applied, Alarm Report Command will be sent to
						Nodes in Grouping 1 to confirm the power applied status for HSM02. 
						Tamper event (17): Grouping 1, afterward, LED will light up for 1 sec to remind user when HSM02 is
						triggered due to open or close incidents. 
						Low battery (255): When HSM02 automatically wakes up, it will check the battery usage. When
						low battery is detected, Alarm Report Command will be sent to Nodes in
						Grouping 1, afterward, LED will light up for 1 sec to remind user when HSM02 is
						triggered due to open or close incidents.
					</Help>
					<Item label="No alarm" value="0" />
					<Item label="Power applied" value="2" />
					<Item label="Tamper event" value="17" />
					<Item label="Low battery" value="255" />
				</Value>
			</CommandClass>
```

Nico0084, 2016-03-25:
What do think about this proposal ?
Too risky compared to potential issues in xml configuration ?

Fishwaldo, 2016-04-11:
I think thats a good idea... we have the xml testing scripts (`make xmltest`) so it should confirm the XML syntax.

Could you open a PR for this?

Fishwaldo, 2016-04-27:
Closing as we merged the PR already.

closed 2016-04-27 by Fishwaldo