OpenZWave/open-zwave #788: Domoticz OpenZWave: Driver Failed!! raspberry pi 3

onsmam, 2016-03-04:
try-ed to upgrade to pi 3 and get the following error:

Error: OpenZWave: Driver Failed!!

try-ed the new stick in pi 2 with razberry, this works, so only difference is the Pi 3, and its hardware...
try-ed to re-install openzwave with no succes. still OpenZWave: Driver Failed!!

gizmocuz, 2016-03-05:
This has absolutely nothing to do with OZW.
If you had searched the Domoticz forum before, or even the internet, you could have found out that there is an issue with the RPI3, and razberry

onsmam, 2016-03-05:
if you had checked this issue sooner than you had noticed that the answer was not available the time of writing...

closed 2016-03-05 by onsmam

gizmocuz, 2016-03-05:
When reporting problems with Domoticz, it might be an idea to report this on the Domoticz forum first?

onsmam, 2016-03-05:
OpenZWave: Driver Failed!! might have something to do witch openzwave..... en at the same time the question was asked... maybe here some one with the same problem.. 

thank you for the kind support....

gizmocuz, 2016-03-05:
?

jdagerot, 2016-04-07:
It would have been nice to redirect new readers on where to look for a solution, I've put a fair amount on trying to find anything about this problem but can't find anything when googling, I'm just getting back on this page.

gizmocuz, 2016-04-08:
Okey, let me put it like this, the next time you have problems with Domoticz/OZW please contact Microsoft first, because it is probably their default USB driver that has issues.