OpenZWave/open-zwave #789: No Neighbors at Z-Stick levels and Z-Stick not reported by Z-nodes

KiwiHC16, 2016-03-04:
I tried with control-panel and Jeedom software and each time i got the same issue: "No Neighbors at Z-Stick levels and Z-Stick not reported by Z-nodes"

SW level master git

I tried many different manipulation switch off,on, hard reset, SW reset,...

zwave log:
2016-02-19 15:19:01.735 Detail, Node001, QueryStage_Neighbors
2016-02-19 15:19:01.735 Detail, contrlr, Requesting routing info (neighbor list) for Node 1
2016-02-19 15:19:01.736 Detail, Node001, Queuing (WakeUp) Get Routing Info (Node=1): 0x01, 0x07, 0x00, 0x80, 0x01, 0x00, 0x00, 0x03, 0x7a
2016-02-19 15:19:01.736 Info,
2016-02-19 15:19:01.736 Detail, Node001, Queuing (WakeUp) Query Stage Complete (Neighbors)

No Neighbors.

Log when Neighbors request:
2016-02-19 16:12:13.551 Detail, Node001, Queuing (Controller) Request Node Neighbor Update
2016-02-19 16:12:13.551 Info, Requesting Neighbor Update for node 1
2016-02-19 16:12:13.551 Detail, 
2016-02-19 16:12:13.551 Detail, Node001, Queuing (Controller) Request Node Neighbor Update
2016-02-19 16:12:13.551 Detail, Notification: ControllerComand - Starting
2016-02-19 16:12:13.552 Detail, Notification: ControllerCommand - Sleeping

No answer..

At Node level, Neighbors list available with out the stick. (40 nodes, none report the controler: Node 001 ):
2016-02-19 15:19:07.515 Info, Node006, Sending (Command) message (Callback ID=0x00, Expected Reply=0x80) - Get Routing Info (Node=6): 0x01, 0x07, 0x00, 0x80, 0x06, 0x00, 0x00, 0x03, 0x7d
2016-02-19 15:19:07.520 Detail, Node006, Received: 0x01, 0x20, 0x01, 0x80, 0xd6, 0xe0, 0xf1, 0xf7, 0xff, 0xe9, 0x53, 0x70, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x5b
2016-02-19 15:19:07.520 Detail,
2016-02-19 15:19:07.520 Info, Node006, Received reply to FUNC_ID_ZW_GET_ROUTING_INFO
2016-02-19 15:19:07.520 Info, Node006, Neighbors of this node are:
2016-02-19 15:19:07.520 Info, Node006, Node 2
2016-02-19 15:19:07.520 Info, Node006, Node 3
2016-02-19 15:19:07.520 Info, Node006, Node 5
2016-02-19 15:19:07.520 Info, Node006, Node 7
2016-02-19 15:19:07.520 Info, Node006, Node 8
2016-02-19 15:19:07.520 Info, Node006, Node 14

I don't remember having this issue with openzwave with older versions ?
I have the same behavior on a raspberry and on a debian/odroid with 2 different sticks.
Any idea ?

KiwiHC16, 2016-03-17:
I have a second stick and the controller is seen by the nodes and it reports Neighbors.
Same software (Firmware, SW,...), one stick ok, the other one not ok.
Issue in the faulty stick ?

Fishwaldo, 2016-04-11:
I'd say so...

closed 2016-04-11 by Fishwaldo

KiwiHC16, 2016-04-11:
Replaced the old one (faulty) by a ne one and no more issue. So looks like this is the stick.