OpenZWave/open-zwave #790: SetValueRaw not represented in DotNet Wrapper

dechutes, 2016-03-07:
Hello,
I'm not sure this is where to ask this, but I have been trying to add the setvalueraw to the dotnet wrapper and can't get it to work.  Could someone point me in the right direction?

Thank you!

bufferUnderrun, 2016-03-15:
Hi,

I use the dotnet wrapper too and maybe i can help. 
Could you post the code you're trying to add ?

dechutes, 2016-03-16:
Hi bufferUnderrun,
Thanks so much for the offer.  I'm trying to access the:
bool SetValue( ValueID const& _id, uint8 const* _value, uint8 const _length );
in Manager.h

It allows me to send an array of bytes that are not translated.  

I was working on this here:
https://groups.google.com/forum/#!topic/openzwave/vxpUsHtxZj4

but I couldn't figure out the C# to C++ conversion.

Any help would be greatly appreciated!

Fishwaldo, 2017-05-30:
@OpenZWave/openzwave-dotnet-uwp  I assume you will do this?

closed 2017-05-30 by Fishwaldo

dotMorten, 2017-05-30:
Yup. Issue already logged. I'd love to implement it, but need to understand how and when to use it first, so I can test it :-)

dechutes, 2017-05-30:
I can help with that.  The original reason for the request is I have a Quickset door lock and the codes for it require raw values to be sent, it doesn't work with string, int or any other type that is already represented in the the dotnet wrapper.  I am able to view the codes after getting them as string and converting the bytes to values.  The only thing missing right now is how to get the byte values in dotnet to convert to the same in c++.  I have tried reading up on this, but don't understand the whole pointer thing.  It's really just a conversion error right now.  the code already exists in c++ to write it i just can't get a reference to that method to compile.

dotMorten, 2017-05-30:
Let's move the discussion over to  https://github.com/OpenZWave/openzwave-dotnet-uwp/issues/2