OpenZWave/open-zwave #792: Memory Leaks

jmattioni, 2016-03-07:
I've been testing the v1.4 openzwave library debug DLL with an MFC program. You may be aware that there is an issue with MFC causing false memory leaks for static objects in linked DLLs. This is not technically a fault of the openzwave library, but I suggest that all static objects be cleared when the Manager is destroyed to avoid this problem. There's a few that are not. I can provide more info if interested. It's mainly the result of not clearing static std list and map objects..

However, after a number of days of debugging a couple of real memory leaks were discovered in the openzwave library. The leaks are all in the Driver class:

1. The destructor for the Driver class is sending a few notifications about its progress. There are comments in that function that indicate the problems with doing that. Well, the result is that (in my case) the last two notifications never get processed and are not destroyed. I left the very first notification (that driver is shutting down) and that apparently gets processed OK. But I suggest removing ALL notifications from the Driver destructor or putting a suitable wait after the notification to ensure it's processed.

2. The AuthKey and EncryptKey objects are not destroyed on Driver destruction. There's a further problem related to this -- Driver function initNetworkKeys will allocate new AuthKey and EncryptKey objects every time it's called, even if they already exist from a prior call. This will then leak the prior key objects.

Fishwaldo, 2016-03-11:
Please provide a list of the static objects. I don't really use Windows Much, so I'm not able to see this list. 

AuthKey and EncryptKey - Good Catch. I'll queue that up

Also, good catch on the NotifyWatchers not being called on the Destructor and cleaning up.

jmattioni, 2016-03-11:
The extra static var cleanup can be in the Manager destructor. Below is my revised code for ~Manager, which shows the specific static objects that needed cleanup. Technically these are not memory leaks since they will get destroyed when static vars are unloaded. The problem I encountered was with a test Windows application using the MS MFC library. The MFC library does memory leak checks, but there's an issue with controlling the order of static destruction resulting in MFC reporting a lot of false leaks for the OpenZWave library. 

The problem is that real memory leaks can hide among the many false leaks so I wanted a solution. The code change below cleans up static objects so that the real leaks can be more easily found. Static std template objects, like list and map, are the main culprits because they use the heap. I also worked out a way to get the C runtime to show file/line number info for leaks in the OZW library. That's how I zeroed in on the leaks I reported.

Manager::~Manager
(
)
{
	// Clear the pending list
	while( !m_pendingDrivers.empty() )
	{
		list<Driver*>::iterator it = m_pendingDrivers.begin();
		delete *it;
		m_pendingDrivers.erase( it );
	}
	m_pendingDrivers.clear();

	// Clear the ready map
	while( !m_readyDrivers.empty() )
	{
		map<uint32,Driver*>::iterator it = m_readyDrivers.begin();
		delete it->second;
		m_readyDrivers.erase( it );
	}
	m_readyDrivers.clear();

	m_notificationMutex->Release();

	// Clear the watchers list
	while( !m_watchers.empty() )
	{
		list<Watcher*>::iterator it = m_watchers.begin();
		delete *it;
		m_watchers.erase( it );
	}
	m_watchers.clear();

	// Clear the generic device class list
	while( !Node::s_genericDeviceClasses.empty() )
	{
		map<uint8,Node::GenericDeviceClass*>::iterator git = Node::s_genericDeviceClasses.begin();
		delete git->second;
		Node::s_genericDeviceClasses.erase( git );
	}
	Node::s_genericDeviceClasses.clear();

	//// (JDM) MORE CLEANUP!
	//// (JDM) Clear the basic device class list
	while (!Node::s_basicDeviceClasses.empty())
	{
		map<uint8, string>::iterator git = Node::s_basicDeviceClasses.begin();
		Node::s_basicDeviceClasses.erase(git);
	}
	Node::s_basicDeviceClasses.clear();

	//// (JDM) Clear the role device class list
	while (!Node::s_roleDeviceClasses.empty())
	{
		map<uint8, Node::DeviceClass*>::iterator git = Node::s_roleDeviceClasses.begin();
		delete git->second;
		Node::s_roleDeviceClasses.erase(git);
	}
	Node::s_roleDeviceClasses.clear();

	//// (JDM) Clear the device type class list
	while (!Node::s_deviceTypeClasses.empty())
	{
		map<uint16, Node::DeviceClass*>::iterator git = Node::s_deviceTypeClasses.begin();
		delete git->second;
		Node::s_deviceTypeClasses.erase(git);
	}
	Node::s_deviceTypeClasses.clear();

	//// (JDM) Clear the node type class list
	while (!Node::s_nodeTypes.empty())
	{
		map<uint8, Node::DeviceClass*>::iterator git = Node::s_nodeTypes.begin();
		delete git->second;
		Node::s_nodeTypes.erase(git);
	}
	Node::s_nodeTypes.clear();
	//// (JDM) END: MORE CLEANUP!

	Log::Destroy();
}

jmattioni, 2016-03-13:
I have a related issue in Node.cpp to report and thought I'd just tack it onto this issue since it also involves a missing call to NotifyWatchers(). The ~Node destructor also needs to call NotifyWatchers() before exit because the deletion of the 'm_values' ValueStore causes notifications to be queued.

PR #1692 (2019-01-22): Cleans up static variables during Manager destruction.
Incorporate jmattioni's changes to fix #792.
commit ef1e9b7a (2019-01-22): Cleans up static variables during Manager destruction. (#1692)

Incorporate jmattioni's changes to fix #792.
closed 2019-05-03 by Fishwaldo