OpenZWave/open-zwave #793: SCS317 : Class 0x20 value

ksoul78, 2016-03-07:
Hi guys. 

I'm using SCS317 thermostat from Secure and I found strange behaviour with 0x20 command class. 

[According to the manual](https://www.domadoo.fr/fr/index.php?controller=attachment&id_attachment=1084), the Basic command class is mapped to the
thermostat set point, with HEAT as the parameter.

Following parameters are used for SET:

> 0x00 – Enables energy saving mode, temporarily overrides the set temperature with the standby temperature.
> 0xFF – Set comfort mode, temporarily regulates the room temperature to 21 degrees.

GET command provides the following responses:

> 0x00 – Energy saving mode enabled
> 0xFF – Energy saving mode disabled

This is working correctly BUT something strange happens when thermostat is changing to heat or idle (0x42) state because the value is written to the 0x20 instance 0 data[0].val under a binary value overwriting the Energy saving mode mentionned above. 

I'm new to OpenZwave used with Jeedom Home Automation Solution, and did not notice this behavior with my precedent software (z-way if this matters). 

Regards, 
ksoul78

Fishwaldo, 2016-04-11:
Can you try adding:
`<CommandClass id="32" ignoremapping="true" \>`
to config/horstmann/scsc17.xml
and see if that fixes the problem?

Fishwaldo, 2016-04-27:
ping... any update?

ksoul78, 2016-05-24:
I'm going to check asap and give you a feedback. 
Thanks

Fishwaldo, 2016-08-18:
No Feedback. Closing.

closed 2016-08-18 by Fishwaldo