OpenZWave/open-zwave #794: Exception: Manager.cpp:1633 - 101 - Invalid ValueID

polarathene, 2016-03-08:
I realize from past closed issues that this is likely due to third party implementation(OpenZwave/node-openzwave-shared in this case) of the OZW API. I have created an issue 5 days ago [here](https://github.com/OpenZWave/node-openzwave-shared/issues/81).

I am reaching out for advice on how debug this so I can fix it and send out a PR(assuming it's a simple fix to pass valid valueID instead). I have no C++ experience and assume that is the location of the bug.

This bug is only affecting some zwave devices, which I am new to, the devices that seem to trigger this error do appear to be `Gen5`/`Zwave Plus` devices as [this](https://github.com/OpenZWave/open-zwave/issues/509) and [this other issue](https://github.com/OpenZWave/open-zwave/issues/520) bring up. That said I have had some odd behaviours as described in the issue I created.

A possible solution is removing the basic set command which seems to be mentioned in the log prior to the exception error popping up. This is described as a solution in [this closed issue](https://github.com/OpenZWave/open-zwave/issues/447), but I am not familiar with how you go about doing this to verify.

These errors are generated by what the third party package refers to as `node event` events, the device is reporting a value change. The events for values being added/removed/changed do not fire before this error occurs. I believe the problem starts [here](https://github.com/OpenZWave/node-openzwave-shared/blob/1dffedca15dd43882f7391a67b22ef3e0ba3c005/src/callbacks.cc#L289-L291), which calls `zwaveValue2v8Value(value)` which was recently moved to [here](https://github.com/OpenZWave/node-openzwave-shared/blob/69765bba655cde894c730271da6dff25f8d2dc52/src/utils.cc#L229).

Fishwaldo, 2016-03-08:
Node_Event doesn't emit a ValueID, so you can lookup any values on it. It only emits a Event which matches to the value sent in the Basic Class.

As basic is usually mapped to other CC's there is usually not a need to react to Basic CC Updates.

Looking at the other ticket, it also appears that the device is still sleeping after restarting OZW. Until the device wakes up and you receive a Type_NodeQueriesComplete notification, nothing will work as expected. 

You should reconfigure any devices that send a Basic Set message to send the relevant Command Class instead.

closed 2016-03-08 by Fishwaldo

ekarak, 2016-03-08:
In which case I need to amend the code highlighted above, as the code is probably reading garbage from memory :(