OpenZWave/open-zwave #801: Enerwave zwn-rsm2 device shares device id with existing device, would like to add parameter

instantchow, 2016-03-15:
The Enerwave ZWN-RSM2 Smart Dual Relay Switch Module ([pdf](http://enerwaveautomation.com/wp-content/uploads/products/zwnrsm2/zwn-rsm2_ac_092514.pdf)) carries the same product ID and device ID as the ZW-20R. There does not seem to be any parameter collision between these two devices as the ZW-20R has one parameter at index 1 while the RSM2 has one parameter at index 3.

I have added the ZWN-RSM2 singular config parameter to the existing zw20r.xml config file. Perhaps they can coexist?

https://github.com/instantchow/open-zwave/commit/bee97ad45a2ad432bc1cce58b7127ec97d4a8abb

Since I don't own the 20R Dual outlet device I cannot tell you definitively if this will be a problem. Perhaps @chadparry (original contributor for this device) can help.

also somewhat related #141

chadparry, 2016-03-18:
I can test a config with the ZW20R if that will help. I agree that there 
isn't any reason they should have a problem coexisting. But it will be 
confusing for users, because inactive config entries will be published. 
I can't think of any solution to avoid that.

On 03/14/2016 09:55 PM, instantchow wrote:
>
> The Enerwave ZWN-RSM2 Smart Dual Relay Switch Module (pdf 
> <http://enerwaveautomation.com/wp-content/uploads/products/zwnrsm2/zwn-rsm2_ac_092514.pdf>) 
> carries the same product ID and device ID as the ZW-20R. There does 
> not seem to be any parameter collision between these two devices as 
> the ZW-20R has one parameter at index 1 while the RSM2 has one 
> parameter at index 3.
>
> I have added the ZWN-RSM2 singular config parameter to the existing 
> zw20r.xml config file. Perhaps they can coexist?
>
> instantchow@bee97ad 
> <https://github.com/instantchow/open-zwave/commit/bee97ad45a2ad432bc1cce58b7127ec97d4a8abb>
>
> Since I don't own the 20R Dual outlet device I cannot tell you 
> definitively if this will be a problem. Perhaps @chadparry 
> <https://github.com/chadparry> (original contributor for this device) 
> can help.
>
> also somewhat related #141 
> <https://github.com/OpenZWave/open-zwave/issues/141>
>
> —
> You are receiving this because you were mentioned.<br />Reply to this 
> email directly or view it on GitHub:<br /><a 
> href="https://github.com/OpenZWave/open-zwave/issues/801">https://github.com/OpenZWave/open-zwave/issues/801</a>
>

Fishwaldo, 2016-03-21:
Unfortunately having non-existent config values will cause timeouts when OZW starts up, and both config values will be exposed in OZW. Best to bitch to the manufacturer about using duplicate ID's

Fishwaldo, 2016-04-11:
When #141 is done, we can add this.

kentcalero, 2017-02-25:
this seems to still be a problem, I'm experiencing it with zw20r (cannot find in zwave alliance db) and zwn-rsm1. How do we handle different devices sharing the same ID?

nechry, 2017-02-25:
We add a commented entry in database pointing on a specific xml file. You must manually comment and uncomment the two entries. It's a manuel action. There files are only useful at the inclusion, if you dont remove your zwcfg file they dont used anymore. It's a workaround but at the moment it's the only solution we have

kentcalero, 2017-02-25:
That's what I figured. A solution for the future could be an update to manually set the config file for each node. This will also allow for easy updating after the node is added. This should not be a complicated add as it's just a button that performs the load procedure and the specific file name entered by the user.

Kent L. Calero

Sent from the field through my iPhone.

Fishwaldo, 2019-05-13:
Closing this. Not much we can do if a manufacturer messes up

closed 2019-05-13 by Fishwaldo