OpenZWave/open-zwave #804: ACK_TIMEOUT limited to 2 secs

raym777, 2016-03-16:
While troubleshooting a problem with a device in my network, I was trying to increase the value of ACK_TIMEOUT in Defs.h from 1sec to 3sec, ie: 

#define ACK_TIMEOUT **3000**

But when I rebuild I get this:

/home/sam/open-zwave/cpp/src/Driver.cpp: In member function ‘void OpenZWave::Driver::HandleGetSerialAPICapabilitiesResponse(uint8*)’:
/home/sam/open-zwave/cpp/src/Driver.cpp:2447:33: error: large integer implicitly truncated to unsigned type [-Werror=overflow]
   msg->Append( ACK_TIMEOUT / 10 );

It looks like anything over 2 secs is not accepted. 

Suspect this may be a bug?

Fishwaldo, 2016-03-16:
I doubt that's going to help you at all.

This ACK_TIMEOUT only affects the timeout we wait for the controller to acknowledge commands sent back and forwards to the controller. It has nothing to do with remote devices ACK’ing messages. 

Unless you have a seriously messed up serial port or a extremely slow controller this probably won’t help with anything network related.

closed 2016-03-16 by Fishwaldo

raym777, 2016-03-16:
Hi Justin, thanks for the reply. The actual problem I'm trying to address relates to a FLiRS device which is very slow to respond. I was hoping that by increasing the ack-timeout, I could overcome this issue.

Fishwaldo, 2016-03-17:
FLIRS devices should be ok, as they are using a newer ZWave Chip.  It sounds like you might have some devices in your network that don't support beaming if you are having timeout issues there. 

Check the Neighbor list of the FLiRS device you are having issues with and make sure all the neighbors support beaming.

raym777, 2016-03-18:
Thanks again. I've confirmed neighbors support beaming. The device is a Remotec thermostat and just slow to respond in FLIRS mode (even says it will behave this way in the user manual). The fix for me was to increase the MAX_TRIES param from 1 to 3. The device usually responds by the second attempt.

FYI - I think FLIRS support may need some tweaking in OZW (but I could be wrong). For example, I've tested pulling the power temporarily to both a mains connected switch and a FLIRS device. During OZW initialisation, the mains device is instantly recognised as dead and moves onto querying the next node right away. Whereas with the FLIRS device, OZW attempts the query over and over, as defined by the MAX_MAX_TRIES definition (I think). I wonder why OZW doesn't retry in this case.

Just an observation and not a big deal in the whole scheme of things. Happy to provide more details if needed. 

Cheers, 
Ray.

Fishwaldo, 2016-03-21:
FLiRS is handled by the RF layer. We don't have any control over it via the Serial API so not much we can do. Sounds like a "issue" with the device. At least retries work for you.