OpenZWave/open-zwave #807: Error in documentation

Forbinn, 2016-03-21:
Hello,

I found an error in the documentation of the Notification class and more precisely the Notification::Type_ControllerCommand enumeration value. Indeed the documentation said that:
* Notification::GetEvent returns a Driver::ControllerCommand and
* Notification::GetNotification returns a Driver::ControllerState

but in fact and after looking at the code on github directly (https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/Driver.cpp#L5511) I found that:

* Notification::GetEvent returns a Driver::ControllerState and
* Notification::GetNotification returns a Driver::ControllerError

I hope this will help you.

Regards,
Vincent

Fishwaldo, 2016-04-11:
Could you point out where in the Doco you are seeing this? 

I've just checked, and I can't find the references you make.

Forbinn, 2016-04-12:
I found the mistake right here: http://www.openzwave.com/dev/classOpenZWave_1_1Notification.html#a5fa14ba721a25a4c84e0fbbedd767d54aadeff93abdefc102a9798f62eb0f3461

Fishwaldo, 2016-04-25:
Fixed. Thanks

closed 2016-04-25 by Fishwaldo