OpenZWave/open-zwave #808: mark battery device as dead for removing

michapr, 2016-03-22:
If you will have dead battery devices (maybe while testing with different controllers without excluding it before - or if defective device,..) you can not remove the node immediately.

Would be nice to have an option "mark device as dead" so that node could be removed from network immediately. Controllers support it.
In this time it take a long time until such a dead device will be removed.

Maybe issue is duplicated, but didn't found it. I think I'm not the first one who is missing this ;)
If duplicated, then sorry,....

nechry, 2016-03-22:
Yes +1 for this feature if is possible.

My current workaround is stop network edit node remove wakeup CC xml element and restart network then the controller think the device is dead.

michapr, 2017-11-22:
Can this be implemented?
As sample in Z-Way I can mark a device as broken - and then remoce it from controller.
Just now I have in result of testing (and defective devices) a lot of invalid nodes in my list.
Would be really nice if it can be marked and removed.

Thanks!

joydashy, 2018-07-17:
Would be a very handy feature, having a lot of trouble getting rid of ghost devices the controller still considers to be alive, but they are not!

daringer, 2019-05-05:
bump, 
I am the https://github.com/daringer/zwave-core developer, the current goal is to have a modern ozwcp replacement and an easy standalone z-wave configuration frontend. I am also struggling with the removal of dead battery nodes and would highly appreciate any workaround or hack you guys made to remove a node which cannot answer anymore. 

@nechry, could you maybe be a little more precise what you do within the xml file? 

@michapr, any more info about the Z-Way solution, what's their zwave backend? openzwave? any hint where to look to get details of the process they apply?

(FYI, for me I have several nodes which have somehow lost the association with the zwave controller, thus are unreachable and currently not removable)

Fishwaldo, 2019-05-13:
Have you tried the Manager::HasNodeFailed Call? This should get the Controller to check the node.

http://www.openzwave.com/dev/classOpenZWave_1_1Manager.html#a8000a9c5faf089be93daa63e92573fd5

michapr, 2019-05-13:
@daringer : Z-Way is an own certified software solution - I have taken this sample (some years ago :) ) only as reference.
Today many commercial solutions (as sample Zipato, ...) can remove any device from controller, even battery nodes.
Would be really good if this will be possible in OZW, even while testing devices it happen, that device must be deleted.

nechry, 2019-05-20:
@daringer sorry for late answer. Yes this is exactly I perform to mark a Battery operated device as dead. To do this You must stop the network, remove the <CommandClass id="132">....<../> xml-sub-tree within the to-be-deleted node and restart the network.
You can found my helper : https://github.com/jeedom/plugin-openzwave/blob/beta/resources/openzwaved/ozwave/node_utils.py#L178

@Fishwaldo unfortunately the Manager::HasNodeFailed call will never mark a device if they can Sleep. With this hack I remove the capability in the cache. 

@Fishwaldo If we implement this hack in openzwave, the best would be not to have to restart the network and force the taking into account of the suppression of the CC Wakeup.

daringer, 2019-05-25:
@nechry, in fact this is the only solution, which worked for me so far to remove a dead battery device. Out of curiosity: is this intended behavior from the zwave - specification point of view?

anyways, thanks a lot works for me :wave:

nechry, 2019-05-25:
@daringer No, it's more of an interpretation in the library. So this hacking does a simple works-around.