OpenZWave/open-zwave #812: Aeon Door/Window Sensor DSB29

ryanborstelmann, 2016-04-01:
I'm having issues with an Aeon DSB29 with both Home-Assistant and Domoticz: When the applications first boot, they are unable to receive commands from the sensor's open/closed state due to the device being asleep. If I go "wake" the device manually by pressing the alarm sensor three times quickly (per [the manual](http://www.vesternet.com/downloads/dl/file/id/293/product/756/z_wave_aeon_labs_door_window_sensor_manual.pdf), the device begins to work great for the duration of the application run. This isn't a huge deal, except every time I restart to change some config, I have to go around to *every* device and wake it, or the device simply doesn't work.

The device worked flawlessly in SmartThings, but their code is closed-source so I'm not sure what they do to make this work. Anyone have advice of what I can set in openzwave control panel to make this work? was thinking of setting the default wake interval, but not sure what that will do to my battery life.

Thanks!

Fishwaldo, 2016-04-07:
Thats the way that OZW works. See the mailing lists for explainations etc.

closed 2016-04-07 by Fishwaldo

antares2001, 2016-12-09:
Eh - could you please explain this a bit and help perhaps how to fix this? I'm using several of these sensors as well, and it makes no sense to every time manually wake them. In that case I'll have to stick away from OZW.