OpenZWave/open-zwave #814: There is no easy way to disable generating a fat binary on MacOS X

julienw, 2016-04-05:
I'd like to generate a thin binary instead of a fat binary on MacOS X (for static linking with Rust). This is currently not easily possible because the LIBS make variable has more things than just `-arch` and it would be error prone to set it from the outside.

julienw, 2016-04-06:
Actually it looks like this issue prevents from buiding in the newer MacOS X (tested in 10.10.3):
```
➜  open-zwave git:(master) make
CPPFLAGS= /Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Volumes/mozilladev/foxbox/open-zwave/cpp/build/ -
Building OpenZWave Version 1.4-79-g0d716a3
Building tinystr.o
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:11:bad register name `%rdi)'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:12:bad register name `%rax)'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:14:bad register name `%r13'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:16:bad register name `%r12'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:18:bad register name `%rdi'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:19:bad register name `%rbp'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:21:bad register name `%rbx'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:23:bad register name `%rsi'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:24:bad register name `%rsp'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:26:bad register name `%rsi'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:27:bad register name `%rax)'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:29:`__ZN11TiXmlString8nullrep_E(%rip)' is not a valid base/index expression
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:30:bad register name `%rax'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:31:bad register name `%rbx'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:33:bad register name `%rbp)'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:34:bad register name `%rsi'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:35:bad register name `%r13'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:37:bad register name `%r12)'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:38:bad register name `%rbp'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:39:bad register name `%rbx'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:41:bad register name `%rdi'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:43:bad register name `%rsp'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:45:bad register name `%rbx'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:47:bad register name `%rbp'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:49:bad register name `%r12'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:51:bad register name `%r13'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:57:bad register name `%rsi)'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:58:bad register name `%rdi'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:59:`movabs' is only supported in 64-bit mode
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:60:bad register name `%rax'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:61:bad register name `%rax,4)'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:62:bad register name `%rcx'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:63:bad register name `%rdx'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:65:bad register name `%r13'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:66:bad register name `%rax'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:67:bad register name `%rax,%r13)'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:68:bad register name `%r12)'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:69:bad register name `%rbx'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:70:`__ZN11TiXmlString8nullrep_E(%rip)' is not a valid base/index expression
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:71:bad register name `%rsi)'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:75:bad register name `%rsp'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:77:bad register name `%rbx'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:79:bad register name `%rbp'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:81:bad register name `%r12'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:83:bad register name `%r13'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:102:bad register name `%r13'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:104:bad register name `%r12'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:106:bad register name `%rbp'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:108:bad register name `%rdi'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:109:bad register name `%rbx'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:111:bad register name `%rdx'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:112:bad register name `%rsp'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:114:bad register name `%rdi)'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:115:bad register name `%rdi)'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:116:bad register name `%rax'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:118:bad register name `%rdx,%rdx,2)'
[... cut ...]
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:843:bad register name `%rax'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:844:bad register name `%r12'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:846:bad register name `%rdi'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:850:bad register name `%rbp'
/var/folders/zh/bl0hypv16pb3x474nvy2f3yh0000gp/T//ccVhJOxf.s:900:Rest of line ignored. 1st junk character valued 64 (@).
make[1]: *** [/Volumes/mozilladev/foxbox/open-zwave/.lib/tinystr.o] Error 1
make: *** [all] Error 2
```

julienw, 2016-04-06:
Just as a data point: on another 10.10.5 computer, this works for me; but the gcc binary is actually clang there, so that may be the reason it works.

julienw, 2016-04-07:
The issue from https://github.com/OpenZWave/open-zwave/issues/814#issuecomment-206433050 is fixed by using clang instead of gcc (https://github.com/OpenZWave/open-zwave/pull/818).

Fishwaldo, 2016-10-07:
We only support clang on macosx for now. Closing this.

closed 2016-10-07 by Fishwaldo