OpenZWave/open-zwave #815: Remotec ZRC-90 not supported

ehallq, 2016-04-06:
Alexander over at https://groups.google.com/forum/#!forum/openzwave told my that this Commands Class isnt yet supported after posting this from the log. I dont know how he could tell that but i will post this here and hope that someone could solve this :)

2016-04-06 17:17:45.005 Info, Node042, Received Central Scene set from node 42: scene id=1 in 0 seconds. Sending event notification.
2016-04-06 17:17:45.005 Warning, Node042, No ValueID created for Scene 1
2016-04-06 17:17:46.521 Detail, Node042, Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x2a, 0x03, 0x80, 0x03, 0x5f, 0x07
2016-04-06 17:17:46.521 Detail

nechry, 2016-04-06:
See https://github.com/OpenZWave/open-zwave/issues/367

ehallq, 2016-04-06:
So it was removed as a milestone for 1.4, but I can't see that it is set as a milestone for 1.6 like Alexander said. When will 1.6 probably be released? Is it a milestone for 1.6?

Fishwaldo, 2016-04-08:
Duplicate

closed 2016-04-08 by Fishwaldo