OpenZWave/open-zwave #816: Building on MacOS X 10.10 is broken -- needs to use clang instead of gcc

julienw, 2016-04-06:
After hacking around to fix https://github.com/OpenZWave/open-zwave/issues/814 on MacOS X 10.10, we have this additional issue. Looks like building with gcc is not supported when using MacOS X' headers.

See the error here:
```
➜  open-zwave git:(master) ✗ make
CPPFLAGS= /Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Volumes/mozilladev/foxbox/open-zwave/cpp/build/ -
Building OpenZWave Version 1.4-79-g0d716a3-dirty
Building tinystr.o
Building tinyxml.o
Building tinyxmlerror.o
Building tinyxmlparser.o
Building hid.o
In file included from /usr/include/dispatch/dispatch.h:51:0,
                 from /System/Library/Frameworks/CoreFoundation.framework/Headers/CFStream.h:15,
                 from /System/Library/Frameworks/CoreFoundation.framework/Headers/CFPropertyList.h:13,
                 from /System/Library/Frameworks/CoreFoundation.framework/Headers/CoreFoundation.h:55,
                 from /System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDLib.h:27,
                 from /System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDManager.h:29,
                 from /Volumes/mozilladev/foxbox/open-zwave/cpp/hidapi/mac/hid.c:25:
/usr/include/dispatch/object.h:143:15: error: expected identifier or ‘(’ before ‘^’ token
 typedef void (^dispatch_block_t)(void);
               ^
/usr/include/dispatch/object.h:362:3: error: unknown type name ‘dispatch_block_t’
   dispatch_block_t notification_block);
   ^
/Volumes/mozilladev/foxbox/open-zwave/cpp/hidapi/mac/hid.c: In function ‘get_string_property_utf8’:
/Volumes/mozilladev/foxbox/open-zwave/cpp/hidapi/mac/hid.c:285:11: error: variable ‘chars_copied’ set but not used [-Werror=unused-but-set-variable]
   CFIndex chars_copied;
           ^
/Volumes/mozilladev/foxbox/open-zwave/cpp/hidapi/mac/hid.c: In function ‘hid_enumerate’:
/Volumes/mozilladev/foxbox/open-zwave/cpp/hidapi/mac/hid.c:450:11: error: variable ‘len’ set but not used [-Werror=unused-but-set-variable]
    size_t len;
           ^
/Volumes/mozilladev/foxbox/open-zwave/cpp/hidapi/mac/hid.c: In function ‘hid_open_path’:
/Volumes/mozilladev/foxbox/open-zwave/cpp/hidapi/mac/hid.c:705:10: error: variable ‘len’ set but not used [-Werror=unused-but-set-variable]
   size_t len;
          ^
cc1: all warnings being treated as errors
make[1]: *** [/Volumes/mozilladev/foxbox/open-zwave/.lib/hid.o] Error 1
make: *** [all] Error 2
```

I think that always building with `cc` and `c++` instead of `gcc` and `g++` should do the trick on most systems. Do you know why we use cc/c++ on FreeBSD only ? (cf https://github.com/OpenZWave/open-zwave/blob/0d716a3526eea2c23691c90073c3c5895af3bd24/cpp/build/support.mk#L47-L56)

julienw, 2016-04-06:
On some systems gcc is mapped to clang and thus it works fine. I don't really know why some other systems are different yet.

Fishwaldo, 2016-10-07:
We only support clang on mac for now. Closing this.

closed 2016-10-07 by Fishwaldo