OpenZWave/open-zwave #843: Set value to a clock is not possible

db85, 2016-04-12:
Hi,

it's not possible to set a value to the clock.
I found the issue:
In ValueInt.cpp you create a copy and pass it to the constructor.

```
// create a temporary copy of this value to be submitted to the Set() call and set its value to the function param
  	ValueInt* tempValue = new ValueInt( *this );
	tempValue->m_value = _value;

	// Set the value in the device.
	bool ret = ((Value*)tempValue)->Set();
```

In the Clock.cpp  you don't use the copied value (_value) now:

```
bool Clock::SetValue
(
	Value const& _value
)
{
	bool ret = false;

	uint8 instance = _value.GetID().GetInstance();

	ValueList* dayValue = static_cast<ValueList*>( GetValue( instance, ClockIndex_Day ) );
	ValueByte* hourValue = static_cast<ValueByte*>( GetValue( instance, ClockIndex_Hour ) );
	ValueByte* minuteValue = static_cast<ValueByte*>( GetValue( instance, ClockIndex_Minute ) );

	if( dayValue && hourValue && minuteValue )
	{
		if (dayValue->GetItem() == NULL) {
			ret = false;
		} else {
//--> HERE
			uint8 day = dayValue->GetItem()->m_value;
			uint8 hour = hourValue->GetValue();
			uint8 minute = minuteValue->GetValue();

			Msg* msg = new Msg( "ClockCmd_Set", GetNodeId(), REQUEST, FUNC_ID_ZW_SEND_DATA, true );
			msg->SetInstance( this, instance );
			msg->Append( GetNodeId() );
			msg->Append( 4 );
			msg->Append( GetCommandClassId() );
			msg->Append( ClockCmd_Set );
			msg->Append( ( day << 5 ) | hour );
			msg->Append( minute );
			msg->Append( GetDriver()->GetTransmitOptions() );
			GetDriver()->SendMsg( msg, Driver::MsgQueue_Send );
			ret = true;
		}
	}

	...
	return ret;
}
```

The result ist, that always the old values are sended to the device.

I fixed it with following hack. But I'm not a c++ developer so I can't evaluate my code. But it's working

```
bool Clock::SetValue
(
	Value const& _value
)
{
	bool ret = false;

	uint8 instance = _value.GetID().GetInstance();

	ValueList* dayValue = static_cast<ValueList*>( GetValue( instance, ClockIndex_Day ) );
	ValueByte* hourValue = static_cast<ValueByte*>( GetValue( instance, ClockIndex_Hour ) );
	ValueByte* minuteValue = static_cast<ValueByte*>( GetValue( instance, ClockIndex_Minute ) );

	if (dayValue != NULL && _value.GetID() == dayValue->GetID()) {
		dayValue->OnValueRefreshed(((ValueList*) &_value)->GetItem()->m_value);
		dayValue->Release();
		dayValue = (ValueList*)&_value; //HERE the important step
	}else if (hourValue != NULL && _value.GetID() == hourValue->GetID()) {
		hourValue->OnValueRefreshed(((ValueByte*) &_value)->GetValue());
		hourValue->Release();
		hourValue = (ValueByte*)&_value; //HERE the important step
	}else if (minuteValue != NULL && _value.GetID() == minuteValue->GetID()) {
		minuteValue->OnValueRefreshed(((ValueByte*) &_value)->GetValue());
		minuteValue->Release();
		minuteValue = (ValueByte*)&_value; //HERE the important step
	}

	if( dayValue && hourValue && minuteValue )
	{
		if (dayValue->GetItem() == NULL) {
			ret = false;
		} else {
			uint8 day = dayValue->GetItem()->m_value;
			uint8 hour = hourValue->GetValue();
			uint8 minute = minuteValue->GetValue();

			Msg* msg = new Msg( "ClockCmd_Set", GetNodeId(), REQUEST, FUNC_ID_ZW_SEND_DATA, true );
			msg->SetInstance( this, instance );
			msg->Append( GetNodeId() );
			msg->Append( 4 );
			msg->Append( GetCommandClassId() );
			msg->Append( ClockCmd_Set );
			msg->Append( ( day << 5 ) | hour );
			msg->Append( minute );
			msg->Append( GetDriver()->GetTransmitOptions() );
			GetDriver()->SendMsg( msg, Driver::MsgQueue_Send );

			ret = true;
		}
	}

	...
	return ret;
}
```

Fishwaldo, 2016-04-25:
I touch a slightly different approach (using static_cast instead). Could you test and let me know if it works?

paulacchang, 2016-04-28:
I had the same issue, and I can verify that your bug fix solved the problem, thanks!

db85, 2016-04-30:
The problem is not fixed 	completely. You have a copy and paste error

```
			uint8 hour = hourValue->GetValue();
			if (_value.GetID() == hourValue->GetID()) {
				ValueByte const * hourvaluetmp = static_cast<ValueByte const*>(&_value);
				hour = hourvaluetmp->GetValue();
			}
			uint8 minute = minuteValue->GetValue();
			if (_value.GetID() == hourValue->GetID()) { //SHOULD BE minuteValue->GetID()
				ValueByte const * minuteValuetmp = static_cast<ValueByte const*>(&_value);
				minute = minuteValuetmp->GetValue();
			}

```

db85, 2016-04-30:
And another problem: In my code I refreshed all values with dayValue->OnValueRefreshed(day) and so on. This is necessary because if I set first the hour and then the minutes or any other combinations then the first value will be overwritten with the old value. This happens for example with my danfoss thermostat that awakes only every 10 minutes.

kvaster, 2016-12-09:
Issue is still not fixed. OnValueRefreshed is really necessary...

bxlouis, 2017-08-03:
Any news on that?

jjhuff, 2018-01-10:
I just opened a new PR that adds the OnValueRefreshed

PR #1418 (2018-02-19): Refresh clock values upon change
This should fix #843 

When updating, say, Hour and Minute, you had to wait for the change report to return before updating the next field.

Example log:
```
2018/01/09 19:48:30 ZWave: Node 7: Polling 2 values.
2018/01/09 19:48:30 Time: 4:16
2018-01-09 19:48:30.279 Info, Node007, Value::Set - COMMAND_CLASS_CLOCK - Hour - 1 - 1 - 5
2018-01-09 19:48:30.279 Info, Node007, Value::Set - COMMAND_CLASS_CLOCK - Minute - 2 - 1 - 10
2018/01/09 19:48:30 ZWave: Node 7 changed: User 0x81.1.1: Hour: 5 
2018-01-09 19:48:30.280 Info, Node007, Sending (Send) message (Callback I [... cut]
closed 2018-02-19 by Fishwaldo