OpenZWave/open-zwave #845: Kwikset locks constant errors

misterleeman, 2016-04-13:
I have 3 Kwikset locks in my Zwave network which intermittently work (everything else works fine).

I receive constant errors (for example) :

Info, Node024, Sending (Send) message (Callback ID=0x79, Expected Reply=0x04) - DoorL
ockCmd_Set (Node=24): 0x01, 0x0a, 0x00, 0x13, 0x18, 0x03, 0x62, 0x01, 0xff, 0x25, 0x79, 0x3d
Info, Node024, Request RTT 1243 Average Request RTT 1282
Error, Node024, ERROR: Dropping command, expected response not received after 
1 attempt(s)

Any ideas ? I've tried every version of openzwave. I'm currently running 1.4.1

closed 2016-04-13 by misterleeman

nechry, 2016-04-13:
fyi I add a config file for this device:
https://github.com/jeedom/plugin-openzwave/blob/master/ressources/openzwave/config/kwikset/smartcode.xml

we will merge config with ozw normally this week.

misterleeman, 2016-04-13:
Thanks. I grabbed it and gave it a spin.

Errors still happen. Looks like I'm not the only one. Wonder if it's something unique to the Kwikset locks ?

nechry, 2016-04-13:
I don't have the device, so if you can test and valide this cfg.

Fishwaldo, 2016-04-19:
I believe this device is a Listening Device that supports Beaming. In this case, you should check all the Neighbours listed in this device also support beaming.

If a packet travels via a Device that doesn't support beaming in the Mesh Network, then you will get timeouts like above. 

How to test: Turn off devices and try a few times. If error still occurs, move on to the next device.

misterleeman, 2016-04-19:
Is there a list of devices that do support beaming ?

So I would need to move a device that supports beaming closer to the lock ?

There is another lock (identical) not very far away (in fact, its the closest device to it) that works. Wouldn't it beam to the lock ? Or do they not beam to other devices ? Do battery powered devices repeat/beam ?

misterleeman, 2016-04-19:
Ah, this would be

FLiRS Compatibility Issues

Thanks ! I'll check it out.

________________________________
From: Lee Eisen <misterleeman@hotmail.com>
Sent: Tuesday, April 19, 2016 7:35 AM
To: OpenZWave/open-zwave
Subject: Re: [OpenZWave/open-zwave] Kwikset locks constant errors (#845)

Is there a list of devices that do support beaming ?

So I would need to move a device that supports beaming closer to the lock ?

There is another lock (identical) not very far away (in fact, its the closest device to it) that works. Wouldn't it beam to the lock ? Or do they not beam to other devices ? Do battery powered devices repeat/beam ?

Fishwaldo, 2016-04-19:
Use http://openzwave.com/dev/classOpenZWave_1_1Manager.html#a683b74b8945f84084343aee53ed646de (isNodeFrequestListeningDevice – Abbreviated FLiRS) to make sure the device can receive beams and 
http://openzwave.com/dev/classOpenZWave_1_1Manager.html#a17e8eda785e2031793a08312b720144b (isNodeBeamingDevice) to make sure its neighbours can beam.

Most Battery powered devices can’t be a a Beaming Device. 
Some Battery Powered Devices are FLiRS
Most new Main Powered Devices Support Beaming
No Mains powered devices should support FLiRS.

Fishwaldo, 2016-04-19:
Yep!