OpenZWave/open-zwave #846: Devolo door / window sensor 2648 (D-Link derivate)

liber-8, 2016-04-13:
Hi devs,

as with #738 / #752 there's also a door / window sensor from Devolo (http://products.z-wavealliance.org/products/1130) which is rebranded and the same device as DCH-Z110 from D-Link (http://products.z-wavealliance.org/products/1151).

It would be great if you could add it to the manufacturer_specific.xml.

Thanks

Fishwaldo, 2016-04-25:
Could you please post the Type and ID of this device (and which one its a duplicate of) as its not specified in the product manuals.

liber-8, 2016-04-25:
Sure..

This is the entry in manufacturer_specific.xml for the D-Link device:

`<Product type="0002" id="000e" name="DCH-Z110 Door/Window 3 in 1 sensor" config="dlink/dch-z110.xml"/>`

And this is what I see in my logs for the Devolo clone:

```
sensor.unknown_id0175_unknown_type0002_id000e_temperature_2
sensor.unknown_id0175_unknown_type0002_id000e_luminance_2
sensor.unknown_id0175_unknown_type0002_id000e_alarm_level_2
sensor.unknown_id0175_unknown_type0002_id000e_alarm_type_2
binary_sensor.unknown_id0175_unknown_type0002_id000e_sensor_2
sensor.unknown_id0175_unknown_type0002_id000e_access_control_2
sensor.unknown_id0175_unknown_type0002_id000e_sourcenodeid_2
```

Just out of curiosity.. could I insert

`<Product type="0002" id="0175" name="Devolo Door/Window 3 in 1 sensor" config="dlink/dch-z110.xml"/>`

below the D-Link entry to get the Devolo device added / supported or is there some other magic that needs to be done?

Thanks.

nechry, 2016-04-25:
we also need the Develo Manufacturer id. give use the xml element for this node from your zwcfgxxx.xml file

liber-8, 2016-04-25:
```
                <Manufacturer id="175" name="Unknown: id=0175">
                        <Product type="2" id="e" name="Unknown: type=0002, id=000e" />
                </Manufacturer>
```

nechry, 2016-04-25:
ok so I will add this:
```<Manufacturer id="175" name="Devolo">
		<Product type="0002" id="000e" name="DCH-Z110 Door/Window 3 in 1 sensor" config="dlink/dch-z110.xml"/>
    </Manufacturer>```

nechry, 2016-04-25:
I prepare this :
https://github.com/jeedom/plugin-openzwave/commit/693485a6ec8689e8cd2cc481b1875af6e65e6514
we merge this week

liber-8, 2016-04-25:
cool, thanks :-)

liber-8, 2016-04-25:
So maybe it makes sense to move line 205

`<Product type="0005" id="0175" name="Devolo Home Control Radiator Thermostat" config="danfoss/z.xml"/>`

into the Devolo manufacturer element too.

nechry, 2016-04-25:
we have to be sure (proove) this device is related to Manufacturer id 175. do you also have this device?

liber-8, 2016-04-25:
Yes I do.. right now it shows up as:

```
<Manufacturer id="2" name="Danfoss">
  <Product type="5" id="175" name="Devolo Home Control Radiator Thermostat" />
</Manufacturer>
```

But I guess it's because it already has an entry in manufacturer_specific.xml, right?

I can delete that entry, reset my configuration and rescan tomorrow if that helps.

nechry, 2016-04-25:
No, is relative to Manufacturer id="2" name="Danfoss". Theres informations (ids) come from device it self.
Some devices are rebranded with old Manufacturer id

nechry, 2016-06-08:
the jeedom config
https://github.com/OpenZWave/open-zwave/pull/901
are now merge

closed 2016-06-08 by nechry