OpenZWave/open-zwave #847: segment fault when getting zero size user code packet

binli71, 2016-04-13:
When using openzwave controller in security mode connecting to door lock ( security device ), the device could send to controller " User Code Packet is 0". The " User Code Packet is 0 " will trigger following segment fault.
"
[39m[39m2016-04-12 20:00:25.815 Info, Node002, Sending (Query) message (Callback ID=0x01, Expected Reply=0x04) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x02, 0x0a, 0x98, 0x80, 0x2f, 0x16, 0x64, 0x0f, 0x3d, 0xdd, 0xa5, 0x3b, 0x05, 0x01, 0xc5:
[39m[34m2016-04-12 20:00:25.823 Detail, Node002,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
[39m[34m2016-04-12 20:00:25.823 Detail, Node002,   ZW_SEND_DATA delivered to Z-Wave stack
[39m[34m2016-04-12 20:00:26.991 Detail, Node002,   Received: 0x01, 0x05, 0x00, 0x13, 0x01, 0x00, 0xe8
[39m[34m2016-04-12 20:00:26.991 Detail, Node002,   ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)
[39m[39m2016-04-12 20:00:26.992 Info, Node002, Request RTT 3693 Average Request RTT 2958
[39m[34m2016-04-12 20:00:27.042 Detail, Node002,   Received: 0x01, 0x1e, 0x00, 0x04, 0x00, 0x02, 0x18, 0x98, 0x81, 0xbf, 0xee, 0xb3, 0xc0, 0x43, 0x72, 0x76, 0xb6, 0xa4, 0x71, 0x58, 0xe4, 0x36, 0x2f, 0x58, 0x76, 0xa4, 0xe0, 0xac, 0x7d, 0x1d, 0x1b, 0xf8
[39m[39m2016-04-12 20:00:27.043 Info, Raw: 0x98, 0x81, 0xbf, 0xee, 0xb3, 0xc0, 0x43, 0x72, 0x76, 0xb6, 0xa4, 0x71, 0x58, 0xe4, 0x36, 0x2f, 0x58, 0x76, 0xa4, 0xe0, 0xac, 0x7d, 0x1d, 0x1b, 0xf8
[39m[34m2016-04-12 20:00:27.043 Detail, Node002, Decrypted Packet: 0x00, 0x63, 0x03, 0x03, 0x00
[39m[34m2016-04-12 20:00:27.044 Detail, 
[39m[39m2016-04-12 20:00:27.044 Info, Node002, Response RTT 3747 Average Response RTT 3756
[39m[39m2016-04-12 20:00:27.044 Info, Node002, User Code Packet is 0
[39m[34m2016-04-12 20:00:27.045 Detail, Node002, Initial read of value
*** Segmentation fault
Register dump:

 RAX: ffff80f30ff653bf   RBX: 00007f0cf80399c8   RCX: 0000000000000002
 RDX: 0000000000000001   RSI: 0000000000000000   RDI: 00007f0cf009ac40
 RBP: 00007f0cf80397e0   R8 : 00000000ffffffff   R9 : 00007f0cf009ac40
 R10: 0000000000000000   R11: 0000000000000001   R12: 00007f0cf0036d50
 R13: 0000000000000003   R14: 0000000000000000   R15: 0000000000000000
 RSP: 00007f0cf8039718

 RIP: 000000348868fcbe   EFLAGS: 00010202

 CS: 0033   FS: 0000   GS: 0000

 Trap: 0000000e   Error: 00000004   OldMask: 00000000   CR2: 00000000

 FPUCW: 0000037f   FPUSW: 00000000   TAG: 00000000
 RIP: 00000000   RDP: 00000000

 ST(0) 0000 0000000000000000   ST(1) 0000 0000000000000000
 ST(2) 0000 0000000000000000   ST(3) 0000 0000000000000000
 ST(4) 0000 0000000000000000   ST(5) 0000 0000000000000000
 ST(6) 0000 0000000000000000   ST(7) 0000 0000000000000000
 mxcsr: 1f80
 XMM0:  000000000000000000000000000000ff XMM1:  000000000000000000000000000000ff
 XMM2:  000000000000000000000000000000ff XMM3:  000000000000000000000000000000ff
 XMM4:  000000000000000000000000000000ff XMM5:  000000000000000000000000000000ff
 XMM6:  000000000000000000000000000000ff XMM7:  000000000000000000000000000000ff
 XMM8:  000000000000000000000000000000ff XMM9:  000000000000000000000000000000ff
 XMM10: 000000000000000000000000000000ff XMM11: 000000000000000000000000000000ff
 XMM12: 000000000000000000000000000000ff XMM13: 000000000000000000000000000000ff
 XMM14: 000000000000000000000000000000ff XMM15: 000000000000000000000000000000ff

Backtrace:
/lib64/libc.so.6[0x348868fcbe]
/usr/lib64/libopenzwave.so.1.4(_ZN9OpenZWave8UserCode9HandleMsgEPKhjj+0x724)[0x34896657a4]
/usr/lib64/libopenzwave.so.1.4(_ZN9OpenZWave4Node25ApplicationCommandHandlerEPKhb+0xf8)[0x34896b7788]
/usr/lib64/libopenzwave.so.1.4(_ZN9OpenZWave6Driver10ProcessMsgEPh+0x2e9)[0x34896af6c9]
/usr/lib64/libopenzwave.so.1.4(_ZN9OpenZWave6Driver7ReadMsgEv+0x711)[0x34896b0891]
/usr/lib64/libopenzwave.so.1.4(_ZN9OpenZWave6Driver16DriverThreadProcEPNS_5EventE+0x288)[0x34896b2208]
/usr/lib64/libopenzwave.so.1.4(_ZN9OpenZWave10ThreadImpl3RunEv+0x13)[0x3489697b23]
/usr/lib64/libopenzwave.so.1.4(_ZN9OpenZWave10ThreadImpl10ThreadProcEPv+0x9)[0x3489697b39]
/lib64/libpthread.so.0[0x3488e07234]
/lib64/libc.so.6(clone+0x6d)[0x34886e76ad]

Memory map:

00400000-00404000 r-xp 00000000 fc:00 135479                             /root/cpp/example/dimmer/dimmer-control
00604000-00605000 rw-p 00004000 fc:00 135479                             /root/cpp/example/dimmer/dimmer-control
00605000-00626000 rw-p 00000000 00:00 0                                  [heap]
3488200000-3488221000 r-xp 00000000 fc:00 111654                         /lib64/ld-2.20.so
3488421000-3488422000 r--p 00021000 fc:00 111654                         /lib64/ld-2.20.so
3488422000-3488424000 rw-p 00022000 fc:00 111654                         /lib64/ld-2.20.so
3488600000-3488799000 r-xp 00000000 fc:00 111766                         /lib64/libc-2.20.so
3488799000-3488998000 ---p 00199000 fc:00 111766                         /lib64/libc-2.20.so
[... 72 lines cut ...]
#8  0x0000003488e07234 in start_thread (arg=0x7f0cf803a700) at pthread_create.c:313
#9  0x00000034886e76ad in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
(gdb)
"

The patch below could fix this issue:

$ git diff cpp/src/command_classes/UserCode.cpp
diff --git a/cpp/src/command_classes/UserCode.cpp b/cpp/src/command_classes/User
index bb69458..9fc2972 100644
--- a/cpp/src/command_classes/UserCode.cpp
+++ b/cpp/src/command_classes/UserCode.cpp
@@ -264,7 +264,8 @@ bool UserCode::HandleMsg
                                memcpy( data, &_data[3], size );
                                value->OnValueRefreshed( data, size );
                        } else {
-                               value->OnValueRefreshed( 0x00, 1);
+                               size = 0;
+                               value->OnValueRefreshed( (uint8 const*)&size, 1 
                        }
                        value->Release();
                }
(END)

The issue doesn't happen all the time. It happens more often with faster CPU.

binli71, 2016-04-13:
pull request for this issue:

https://github.com/binli71/open-zwave/pull/1

closed 2016-04-25 by Fishwaldo

openzwave-dev-watcher, 2016-04-30:
I am hitting this same error while trying to pair a Schlage BE469 lock in 
security mode. I was wondering if there is a solution or work around. I 
also noticed your comment regarding this doesn't happen all the time but it 
happens with a faster CPU, could using an older slower CPU be a work around?

openzwave-dev-watcher, 2016-04-30:
> You received this message because you are subscribed to the Google Groups
> "OpenZWave" group.
> To unsubscribe from this group and stop receiving emails from it, send an
> email to openzwave+unsubscribe@googlegroups.com.
> For more options, visit https://groups.google.com/d/optout.

This was recently fixed:
https://github.com/OpenZWave/open-zwave/commit/4314523d320f96fd67a8692c3310bd7fedd7bc94

openzwave-dev-watcher, 2016-05-01:
> On Fri, Apr 29, 2016 at 7:09 PM, Photo 848 <phot...@gmail.com
> email to openzwave+...@googlegroups.com <javascript:>.

How can I implement this fix into my configuration? Sorry, bit of a noob 
but I would like to see if this fixes my issue. If I copy this file to my 
folder do I need to recompile?

binli71, 2016-05-01:
Yes. After apply the fix in the source code, it is required to recompile 
to pick up the fix.
Regards

Photo64, 2016-05-01:
I applied the fix to the source code and recompiled however I still receive a segmentation fault when trying to add a lock securely.

(Segmentation fault (core dumped))

binli71, 2016-05-01:
Sounds like the root cause may be different.

Do you have the log ( such as what happened before the seg fault ), and 
trace  back from core dump ( where the seg fault happened )?

Regards

Photo64, 2016-05-01:
I have the log -> http://hastebin.com/ututopevis.rib

binli71, 2016-05-02:
According to the log, it should be the same problem.
It is better to check the patch being applied properly or not. For 
example, add some Log::Write to show patch is being used in the code...
Depending on how you build the load, the source code may be over written 
after you manually applied the patch.

Regards

mnestor, 2016-05-02:
I worked with Photo64 on this and found that I think the size also needs to be set to 0. If it's 1 then we get the segfault, setting to 0 keeps that from happening. As it's going into ValueRaw it looks like it'll still try to copy memory even though there is nothing for it to do.

This is the stacktrace 
[Switching to Thread 0x7ffff5b33700 (LWP 2930)]
0x000000000044f80a in OpenZWave::ValueRaw::OnValueRefreshed(unsigned char const*, unsigned char) ()
(gdb) bt
#0  0x000000000044f80a in OpenZWave::ValueRaw::OnValueRefreshed(unsigned char const*, unsigned char) ()
#1  0x0000000000491943 in OpenZWave::UserCode::HandleMsg(unsigned char const*, unsigned int, unsigned int) ()
#2  0x000000000046b6c2 in OpenZWave::Node::ApplicationCommandHandler(unsigned char const*, bool) ()
#3  0x0000000000466149 in OpenZWave::Driver::ProcessMsg(unsigned char*) ()