OpenZWave/open-zwave #848: Makefile change to support building openzwave in bitbake environment ( Yocto Project )

binli71, 2016-04-13:
I would suggest following change in the 
cpp/build/support.mk
cpp/build/Makefile
in order to build openzwave in bitbake environment as a layer in yocto project.

The patch details:

diff --git a/cpp/build/support.mk b/cpp/build/support.mk
index 3f74c7e..66dc42f 100644
--- a/cpp/build/support.mk
+++ b/cpp/build/support.mk
@@ -43,6 +43,15 @@ endif
 # version number to use on the shared library
 VERSION := $(VERSION_MAJ).$(VERSION_MIN)
 
+# using setting from bitbake
+ifeq ($(BITBAKE_ENV),1)
+CC     := $(CC)
+CXX    := $(CXX)
+LD     := $(CXX)
+AR     := $(AR)
+RANLIB := $(RANLIB)
+else
+
 # support Cross Compiling options
 ifeq ($(UNAME),FreeBSD)
 # Actually hide behind c++ which works for both clang based 10.0 and earlier(?)
@@ -61,6 +70,8 @@ else
 AR     := $(CROSS_COMPILE)ar rc
 RANLIB := $(CROSS_COMPILE)ranlib
 endif
+
+endif
 SED    := sed
 
 
@@ -96,9 +107,15 @@ else
 pkgconfigdir ?= $(shell pkg-config --variable pc_path pkg-config | awk '{split(
 endif
 
+ifeq ($(BITBAKE_ENV),1)
+sysconfdir := $(PREFIX)/etc/openzwave/
+includedir := $(PREFIX)/include/openzwave/
+docdir := $(PREFIX)/share/doc/openzwave-$(VERSION).$(VERSION_REV)
+else
 sysconfdir ?= $(PREFIX)/etc/openzwave/
 includedir ?= $(PREFIX)/include/openzwave/
 docdir ?= $(PREFIX)/share/doc/openzwave-$(VERSION).$(VERSION_REV)
+endif
 
 top_builddir ?= $(CURDIR)
 export top_builddir
diff --git a/cpp/build/Makefile b/cpp/build/Makefile
index a632f79..31af00a 100644
--- a/cpp/build/Makefile
+++ b/cpp/build/Makefile
@@ -93,6 +93,10 @@ else
 hidapi := $(notdir $(wildcard $(top_srcdir)/cpp/hidapi/linux/*.c)) # we do not 
 endif
 
+ifeq ($(BITBAKE_ENV),1)
+ar_option := "rc"
+endif
+
 cclasses := $(notdir $(wildcard $(top_srcdir)/cpp/src/command_classes/*.cpp))
 vclasses := $(notdir $(wildcard $(top_srcdir)/cpp/src/value_classes/*.cpp))
 pform := $(notdir $(wildcard $(top_srcdir)/cpp/src/platform/*.cpp)) \
@@ -139,7 +143,7 @@ $(LIBDIR)/libopenzwave.a:   $(patsubst %.cpp,$(OBJDIR)/%.o,$
                        $(patsubst %.cpp,$(OBJDIR)/%.o,$(indep)) \
                        $(OBJDIR)/vers.o
        @echo "Linking Static Library"
-       @$(AR) $@ $+
+       @$(AR) $(ar_option) $@ $+
        @$(RANLIB) $@
 
 $(LIBDIR)/$(SHARED_LIB_NAME):  $(patsubst %.cpp,$(OBJDIR)/%.o,$(tinyxml)) \

The sample to compile and install openzwave in bitbake ( yocto )

do_install() {
	export PREFIX="${D}${prefix}"
	export pkgconfigdir="${PKG_CONFIG_DIR}"
	export instlibdir="${D}${prefix}${base_libdir}"
	make BITBAKE_ENV=1 install
}

do_compile() {
	export PREFIX="${D}${prefix}"
	export pkgconfigdir="${PKG_CONFIG_DIR}"
	export instlibdir="${D}${prefix}${base_libdir}"
	make BITBAKE_ENV=1
}

binli71, 2016-04-13:
pull request for this topic:

https://github.com/OpenZWave/open-zwave/pull/849
https://github.com/OpenZWave/open-zwave/pull/850

Fishwaldo, 2016-04-25:
We will deal with this in the PR.

closed 2016-04-25 by Fishwaldo

modjo756, 2017-06-29:
Just to inform you that i make a recipe for ozm [here](https://github.com/modjo756/meta-zwave/blob/master/recipes-ozw/openzwave/openzwave_1.4.bb) with the help of your post